LongSoft/UEFITool #137: UEFIReplace bug

LSSoniX, 2018-06-19:
Всем доброго здравия.

Обнаружилась проблема при замене файлов в PEI томах.
Используется комманда
uefireplace bios.bin 17088572-377F-44EF-8F4E-B09FFF46A070 1 mcode.ffs -o bios.bin -asis
После замены всех файлов и прошивки получаем кирпич.
Сообщений об ошибках при замене нет.
Сделал аналогичную замену испольщуя MMTool, всё работает
Вот архив с файлами
https://cloud.mail.ru/public/AQLL/XcaaDvsuv
org_bios.bin исходный файл
mcode.ffs файл для замены
25_bios_ur файл после обработки УР
25_bios_mmt,bin файл после обработки ММТул
Если сравнить оба файла 25_, то разница только в одном месте 7FFFE0/E1
Вот тут, по моей просьбе товарищ посмотрел и отписал, что не так
https://forums.overclockers.ru/viewtopic.php?p=15630217#p15630217
Также я проверил замену вручную в UEFITool 0.25,
Менялись последние 2 файла, первый "пустышка", "как есть" и всё прошло хорошо, УТ перситал адрес.
Если есть возможность, то пожалуйста поправьте этот баг в УР.
Спасибо.

LSSoniX, 2018-06-22:
Вспомнил, что у асус 9 серии GOPDriver тоже располагается в PEI томах где и файлы микрокодов.
Арихив stuff1 по ссылке, что указал в предсообщении
Размещение в GUID 5BBA83E6-F027-4CA7-BFD0-16358CC9E123
Комманда на замену
`uefireplace bios.bin 5BBA83E6-F027-4CA7-BFD0-16358CC9E123 10  IntelGopDriver.efi -o bios.bin -all`
Замена проходит без ошибок и при этом адрес в 0x7FFFE0 пересчитывается как надо.
Получается, что когда добавили поддержку замены FFS RAW тип секции 1 и "asis" про пересчет забыли. :)

vit9696, 2018-06-22:
Привет, я тебя услышал и что-то даже воспроизвёл, но пока не было минутки посмотреть подробно что там не так, и должно ли оно вообще работать таким образом. "Пересчёта" как такового там нет нигде, и странно, что с UEFITool работает, возможно зависимости где-то криво помечаются, но это я посмотрю ближе к выходным.

LSSoniX, 2018-06-22:
Ну я просто проверял другие варианты, с другими секциями, вот и поделился. :)

vit9696, 2018-06-22:
А подкинь этот GOP драйвер и получившиеся файлы от uefireplace/uefitool/mmtool.
И ещё, я позавчера пробовал получить с помощью uefitool что-то похоже на два файла выше (после uefireplace и mmtool), и у меня получались сильно отличающиеся результаты. Так что прикрепи заодно то, что у тебя с uefitool выходит.

LSSoniX, 2018-06-22:
Лови https://cloud.mail.ru/public/AQLL/XcaaDvsuv
Залил всё в одно.
Z87 там замена только микодов в послдених двух томах "as us"
org_ исходный
ur_ полсе UEFIReplace - не работает
ut_ после UEFITool - работает
ur_ut_ после UR затем UT - не работает
mmt_ после MMTool работает

Z97 замена только GOP в последних двух томах
org_ исходник
ur_ полсе UR
ut_ росле UT
mmt_ после ММТ, тк ммт не работает с секциями, то замена делалась FFS "as is"
Проверить работоспособность не могу, тк плат 9-ой серии у меня нет, но могу точно заверить, что все варианты рабочие.

vit9696, 2018-06-22:
Так, стоп. Вот ты говоришь, что UEFITool заменяет только последние два микрокода, а UEFIReplace все три. Это неравноценный тест, что происходит при замене всех трёх элементов через UEFITool, как это делает UEFIReplace.

LSSoniX, 2018-06-22:
Смотри 3_ut_bios.bin
Первый файл это "пустышка" и он находится в DXE томе.
ММТ тоже меняет 3 файла, при сборке я ремарил по очереди сначала из оригинала сборка в УР, потом из оригинала через ММТ.
Сейчас я сделал тоже самое в УТ.
Те выташил "пустышку" из DXE, забил все 1708... сторонними файлами.
Затем заменил 2 последних в PEI, сохранил и затем вернул "пустышку" в DXE.

vit9696, 2018-06-22:
Что? Мы видимо друг-друга не понимаем.
uefireplace bios.bin 17088572-377F-44EF-8F4E-B09FFF46A070 1 mcode.ffs -o bios.bin -asis
Заменить три объекта, в UEFITool можно также взять и заменить все три объекта. Работать будет?

LSSoniX, 2018-06-22:
Если заменит УР, то работать не будет, а если УТ то будет.
Сравни файлы ur_bios и ut_bios, обычным хексом/
Если я скорректирую ur_bios в 0x7FFFE0, как ut_bios, то заработает.

LSSoniX, 2018-06-22:
Извини, вот все файлы участвующие в замене stuff_4, по той же ссылке.
Приведу упрощенный вариант, тк мы знаем, что где лежит.

```
#извлекаем файл на проверку "пустышка" или нет
uefiextract org_bios.bin 17088572-377F-44EF-8F4E-B09FFF46A070 -o tmpr -m file
#тк мы уже знаем, что это "пустышка", то сразу меняем в DXE томе
uefireplace org_bios.bin 17088572-377F-44EF-8F4E-B09FFF46A070 1 z_cpu0.ffs -o bios.bin -asis
#дальще мы уже знаем, что оставшие 2 файла не "пустышки" и они должны обновиться смкодаии в обоих PEI томах
uefireplace bios.bin 17088572-377F-44EF-8F4E-B09FFF46A070 1 z_cpu1.ffs -o bios.bin -asis
uefireplace bios.bin 17088572-377F-44EF-8F4E-B09FFF46A070 1 z_cpu1.ffs -o bios.bin -asis
#мы заменили все таким образом все GUID, сторонними файлами
#теперь возвращаем уже всё на место, сначала микоды
uefireplace bios.bin 17088572-0000-0000-0000-000000000002 1 mcode.ffs -o bios.bin -asis
uefireplace bios.bin 17088572-0000-0000-0000-000000000002 1 mcode.ffs -o bios.bin -asis
#мы вернули микоды и теперь восстанавливаем ранее извлеченную "пустышку"
uefireplace bios.bin 17088572-0000-0000-0000-000000000001 1 tmpr\file.ffs -o bios.bin -asis
#процедура обновления завершена.
```
Можно всё тоже самое проделать и в УЕФИТул, разница только в том, что в УТ "пустышку" можно не трогать, она никак не влияет на конечный процесс замены.
Тоже самое можно проделать с мумутул, вместо реплейсера, комманда только попроще
'mmtool bios_image  /r GUID file_ffs'
И как я указал ранее, что если замену делать в УТ или ММТ, то файлы на выходе одинаковые, апосле УР не совсем в одном местечке. :)
Извини, если опять что то не так пояснил.

vit9696, 2018-06-22:
Теперь понятно. Вот сразу бы так. Багу можно проявить UEFITool'м. Заменить пустышку и первый микрокод. Получится кирпич и полностью идентичный файл с:
```
uefireplace org_bios.bin 17088572-377F-44EF-8F4E-B09FFF46A070 1 z_cpu0.ffs -o bios.bin -asis
uefireplace bios.bin 17088572-377F-44EF-8F4E-B09FFF46A070 1 z_cpu1.ffs -o bios.bin -asis
```
Проблема в том, что при замене файла в одном из томов не происходит reconstruct с последующим rebase следующего. В случае использования UEFITool ты rebuild ставишь руками (на следующие тома), а вот в случае UEFIReplace просто ломается том.

Я попробовал это пофиксить, и на твоих файлах у меня всё сработало. Тем не менее, это сильно меняет поведение всех утилит, так что крайне желательно проверить на большом количестве биосов. Фигово, что в UEFITool нет автотестов. Эх.

LSSoniX, 2018-06-23:
Ну я пока вчеращний выпуск, на своих Z87, прооверил замену и прошку - всё работает.
Другие платы токо выложить на овере да вин-райд, бантик я уже подготовил, но если есть мысли то задержу тестовый выпуск.
Сразу могу точно сказать, что от владельцев плат на х299 пололожительных отзывов можно не ждать, особенно если это платы от MSI у них даже мумутул и УТ отказываются менять FFS в PEI, хотя реплейсер менят, но с предупреждениями. Остальные бренды на х290 еадо наблюдать.
Зато на остальных всё должно быть гладко.

А по поводу автотеста думаю, что можно взять на заметку для УТ на новом движке.

vit9696, 2018-06-23:
Привет, да, пока не торопись. Лучше сделать один раз нормально, чем с такими последствиями.

vit9696, 2018-06-23:
Попробуй последний вариант. Он должен быть чище и лучше.

LSSoniX, 2018-06-23:
Привет.
Да уже, последнюю сборку от 20:01, на своих проверяю.
Замена без ошибок, после прошивки всё работает.

LSSoniX, 2018-06-24:
Привет.
Всё подготовил для, если изменений больше не предвидится, то выпускаю на тестирование.

vit9696, 2018-06-24:
Ну давай я релиз сделаю, и будет норм.

LSSoniX, 2018-06-24:
Упс. А куда теперь отписать результаты?

vit9696, 2018-06-24:
Ну, если баг будет, то он будет уже новым, я полагаю? Если же этот недофиксили, то переоткрывай. Давай оптимистично допустим, что мы всё починили :D

LSSoniX, 2018-06-24:
Я про х299 как то говорил, с ними еще не рассматривали, тк плат у нас нет, но походе есть тестер на вин-райде. У него асус на х299.
Пока результаты такие:
- Реплейсер - кирпич
-- мумутул - всё работает
при этом файлы после замен сильно отличаются
Сейчас я ему собрал в УЕФИТул и жду ответа.

LSSoniX, 2018-06-25:
Всем доброго здравия.
vit9696, как я и ожидал, тестер отписался, что кирпич с POST кодом 55, если память не имзменяет Memory not instaled по книжке асус. У тестера ROG-RAMPAGE-VI- EXTREME на х299 чипсете.
Я забыл сразу сравнить результаты замены микодов через УР и УТ, а они оказались идентичны.
В общем ситуация такова:
ROG-RAMPAGE-VI-EXTREME-ASUS-1401\
процедура обновления как обычно
по результатам получается, что
УР и УТ дают кирпич
ММТул всё работает.
Я сравнил файлы ur_bios и mmt_bios и увидел, что ММТ радикально патчит содержимое внутри ТЕ секций в PEI тома, а УР и УТ оставляют их такими же как и в исходном org_bios.
Странно, откуда ммт знает, что и где надо конкретно пропатчить, да еще и в куче файлов,  То лм база внутри ммт заложена, то ли ссылки какие перехватывает, непонятенько.

vit9696, 2018-06-26:
Ну, создавай issue? Когда работало, когда не работало, файлы прикладывай, команды… Что ты как маленький. С текстом без ничего я работать же не могу, правда?

platomav, 2018-06-28:
Based on [this](https://www.win-raid.com/t785f16-Discussion-UBU-Tool-related-Questions-Reports-and-Suggestions-272.html#msg54189) report, [here](https://mega.nz/#!vFFXDI5K!g_BNb0iTiZwX8xL51ss_2ezoKX2okWtSTjictu2L6k4) are the results. The reproduction steps should be the same as [here](https://github.com/LongSoft/UEFITool/issues/137#issuecomment-399459003) or [here](https://github.com/LongSoft/UEFITool/issues/137#issuecomment-399590380).

ROG-RAMPAGE-VI-APEX-ASUS-1401.bin --> Stock
ROG-RAMPAGE-VI-APEX-ASUS-1401_UEFIReplace.bin --> UEFIReplace 0.25.1 (Brick)
ROG-RAMPAGE-VI-APEX-ASUS-1401_MMTool.bin --> MMTool (Works)

Some comparison pictures which might be of interest. All of them concerning the GUID after the one we replaced. Same for the next volume. On the left we have UEFIReplace (brick) and on the right MMTool (ok).

[image]
[image]
[image]

LSSoniX, 2018-06-28:
Данная проблема на всех платах х299 от Асус. Как я уже говорил, ммтул сильно патчит содержимое в TE Image Section в PEI томах.

vit9696, 2018-06-29:
Ok, will look at it later.
Хорошо, будем посмотреть.

LSSoniX, 2018-07-22:
Всем доброго здравия.
[image]
Данная проблема (окирпичивание) всплыла в последнее время при замене микрокодов в PEI томах у пользователей ASRock и MSI еа платфореме Аптио 4 на платах 7-9ых сериях чипсетов.
Опять же ммтул справляется с ними без проблем, Я сранил результаты, ммт патчит уже не ТЕ файлы (их там нет), а содержимое PE32 файлов.

Поэтому хочу спрсить, есть возможность реализовать, чтобы при таком сообщении замена файла не производилась и возвращался не нулевый errorlevel? До выяснения вснх обстоятельств с ребилдом PEI

dsanke, 2020-09-29:
This commit causes the problem that pei volume destoryed when replacing microcode in non-pei volume
destoryed https://github.com/LongSoft/UEFITool/commit/35c207dd8c31215fe11b56dbe2542674ccee6b6b
I found the key to bypass these codes without compiling so that I can use v0.25.1 and above to replace microcode
UEFIPatch/Replace 0.25.1 win32
E9 4F FE FF FF 85 C0 0F 84
UEFIPatch/Replace 0.28.0 win32
C6 7D AE 8B 45 A4 85 C0 0F 84
Change 85 C0 to 33 C0

NikolajSchlej, 2022-09-13:
I'm mass-closing all issues that are specific to the old engine, because no further work is planned to be done on it, and all the bugs that are present there are to remain there forever. The original engine had a good run, but it needs to go in peace.

closed 2022-09-13 by NikolajSchlej