LongSoft/UEFITool #139: Uefi could not turn on my bios warning: invalid uefi volume;

947614975, 2018-07-17:
My computer manufacturer restricted the BIOS, which prevented me from booting Android and unlocking TDP. 
Uefitool could not turn on my bios warning: invalid uefi volume; 
factory bios; it doesn't look like a problem, but there are no options 
my modified setup configuration BIOS via AMI BCP,But flash into the system doesn't change ; even if I turn off VT it  No effect; but my logo can be modified through changlog
[image]

[Uploading ezbook-3sl.zip…]()
[image]

947614975, 2018-07-17:
[image]

947614975, 2018-07-17:
[image]

947614975, 2018-07-17:
[image]

vit9696, 2018-07-17:
Could you upload your BIOS dump please? Looks like you forgot :)

947614975, 2018-07-17:
GITHU Bwarning：Something went really wrong, and we can’t process that file.  
So I couldn't get my last my BIOS
So I used mega.
this is factory bios
https://mega.nz/#!bNViUKQK!eHfSU55JTG5CRdO7ZITYkwNnOUdvAmgn3ilqtw47l7I
this is Modificationsetup configuration
https://mega.nz/#!TJEy0QaY!wW-3hdvxEfW8mMTYD8mq240mxGh8UCQ2SoZyDL0IR-E

Thanks

vit9696, 2018-07-17:
Your roms appear to have intersecting regions. Here is a fixed one:
[afuwin_20180714050901_fixed.rom.zip](https://github.com/LongSoft/UEFITool/files/2203069/afuwin_20180714050901_fixed.rom.zip)

LSSoniX, 2018-07-17:
@vit9696 
Приветствую. Вся проблема в использовнии AMIBCP. Для Аптио 5 данная софтина только портит биосы, уже проверено многократно, одни кирпичи получались. Советуем держаться от данной утилиты поодальше

vit9696, 2018-07-17:
Привет, да, знаю, надо работать над NE.

platomav, 2018-07-19:
As @LSSoniX said, the SPI regions gets corrupted due to a buggy AMIBCP version. Thus, this is not an UEFITool issue. Unlocking the TDP via IFR or similar is also not UEFITool related. At best, it is modding related.

closed 2018-07-21 by vit9696