LongSoft/UEFITool #140: Rebuild PEI module creates invalid TE

lasalvavida, 2018-07-18:
Firmware for [SuperMicro A2SDi-2C-HLN4F](https://www.supermicro.com/products/motherboard/atom/A2SDi-2C-HLN4F.cfm) (can be downloaded from this page under "update your bios")

The PEI modules contain TE images. Performing a rebuild operation on any of them without making any other changes will create an invalid TE and the firmware will no longer boot properly.

Attached is an example using ProjectPei (B873591A-E633-4CFD-99F7-48D13207D107)

[pei-modules.zip](https://github.com/LongSoft/UEFITool/files/2207167/pei-modules.zip)

[image]

lasalvavida, 2018-07-18:
Uncommenting the code here https://github.com/LongSoft/UEFITool/blob/master/ffsengine.cpp#L4012 (and adding a close brace) resolves this issue.

Looks like there are no plans to fix this in the old engine, but figure it may be useful for somebody with the same issue.

dsanke, 2018-08-03:
I found the same problem when modifying modules that the volume coutains some pad-file and PEI modules . It will remove some ( not all ) empty pad-file and cause the address in PEI become incorrect .

NikolajSchlej, 2022-09-13:
I'm mass-closing all issues that are specific to the old engine, because no further work is planned to be done on it, and all the bugs that are present there are to remain there forever. The original engine had a good run, but it needs to go in peace.

closed 2022-09-13 by NikolajSchlej