LongSoft/UEFITool #145: UEFIExtract и UEFIReplace предложение новой функции

LSSoniX, 2018-09-26:
Всем доброго здравия!

Порой приходится сталкиватся к примеру вот с такой ситуацией,
[image]
В данном случае UE извлекает всё что есть, а UR заменяет либо первый по списку либо всё подряд, что согласитесть не есть хорошо.
В результате возникло предложение расширить возможности данных утилит еще одной опцией [-p pattern]
`UEFIEctract <BIOS file> <GUID\|SubGUID> [-o folder] [-m body\|file\|\|all] [-t secton type] [-p pattern]
UEFIReplace <BIOS file> <GUID\|SubGUID> <File content> <-t secton type> [-o folder] [-all] [-asis] [-p pattern]`
Суть новой опции в том, что утилиты будут производить действия пользователей только с тем файлом в заданном GUID}SubGUID в котом присутствует искомый паттерн.

Я понимаю, что реализация не так проса, но возьмите на заметку. Думаю, что многим эта идея придется по вкусу.

Спасибо. :)

vit9696, 2018-09-30:
В принципе код для паттернов уже есть, можно было бы добавить. Но есть одно но, сейчас я пытюсь, к сожалению, это наиближайшее к реальности слово, найти время заняться UEFITool NE, чтобы в ней работал билдер, и соответственно все утилиты.

Со старой кодовой базой работать уже не хочется. Недавно нам прислали патч извне, и сейчас в исходниках NE билдер есть, а утилиты я даже частично переписал. Однако при детальном рассмотрении стало понятно, что по существу билдер целиком и полностью сломан, и его только писать с нуля.

Я постараюсь что-нибудь придумать по этой и другой проблеме, но пока всё застопорилось, так как времени катастрофически не хватает.

LSSoniX, 2018-10-03:
@vit9696 
Понимаю. Лучше новым заняться, чем старый продолжать пилить. :)

LSSoniX, 2018-11-11:
Всем доброго здравия.

Появилось еще вот такое предложение.
UEFIExtract (УЕ) мзвлекает нужный файл|тело по заданному GUID, который мы либо знаем заранее либо получаем от UEFIFind (УФ).
Я думаю, что можно было бы сократить процедуру УФ -> УE и сразу задать паттерн в УЕ. То есть, обычно используется сначала УФ с паттерном для поика GUID, а потом этот GUID передаем в УE. Но если я правильно понимаю, то поском паттерна может заняться и УЕ?
К примеру
'uefiextract pattern <-p pattern> -o folder -m file|body'
УЕ сам найдет все паттерны и извлечет в указанную папку файлы .

Этим я не предлагаю отказаться от УФ, он нужный и очень, просто по некоторым нюансам мы сократим время поиск - извлечение.

Спасибо.

LSSoniX, 2018-11-11:
Дополнение.
Я к чему веду.. Для аргумента <-m>, проблема в том, что для более точного извлечения или замегы  пользователю надо указывать тип секции, но это не всегда возможно указать ее, так как бывает, что в одном бивисе это RAW, в другом FreeFotm и тд.
Указатель паттерна <-p> мог бы являться альтернативой <-t>, как один из вариантов возсожностей утилит. А сочетании <-m -t -p> еще более точным ппри работе с указанным в первом сообщении.
Не сочтите за занудство и очень ваи благодарен. :)

LSSoniX, 2018-11-12:
@vit9696
Проверил текущиую сборку, всё хорошо. Спсибо. :)

Теперь по пустой папке при ошибке в аргументах
Бивис для проверки https://cloud.mail.ru/public/64Lc/8vuiN6yVU
За пример возьмем извлечение ЕФИ ИРСТ., секция PE (-t 10)
И так
Извлекаем FFS как есть
`uefiextract bios.bin 91B4D9C1-141C-4824-8D02-3C298E36EB3F -o IRST_f -m file`
Папка с FFS и код 0

Делаем ошибку в GUID
`uefiextract bios.bin 91B4D9C1-141C-4824-8D02-3C298E36EB3A -o IRST_f -m file`
Папки нет, код 1

Извлекаем все тела
`uefiextract bios.bin 91B4D9C1-141C-4824-8D02-3C298E36EB3F -o IRST_b -m body`
Есть папка сбинарниками и код 0
Ошиька в GUID
`uefiextract bios.bin 91B4D9C1-141C-4824-8D02-3C298E36EB3A -o IRST_b -m body`
Ничего нет и код 1

В этих строках всё хорошо. Тепреь уже используеи аошуиент <-t>
Правильная строка
`uefiextract bios.bin 91B4D9C1-141C-4824-8D02-3C298E36EB3F -o IRST_b_t10  -m body -t 18`
Пакпка, файлик и код 0.

Теперь путаем GUID
`uefiextract bios.bin 91B4D9C1-141C-4824-8D02-3C298E36EB3A -o IRST_b_t10  -m body -t 18`
Ничего нет и код 1.
Путаемся в указании Секции. То что нам надо это -t 10, но ошибёмся
`uefiextract bios.bin 91B4D9C1-141C-4824-8D02-3C298E36EB3F -o IRST_b_t18  -m body -t 18`
Секции 18 в FFS нет, но папка рисуется пустая и код 0.
И подобный фокус везде где указан не правильный тип секции.

PS Хочу попросить, глянуть, может удастся хотя бы в качестве скрытой тестовой опции добавить возможность извлекать FFS (-m file), если есть Секция Compessed, извлекать как есть, но в сжатом виде, пока курится #150 

Спасибо.

vit9696, 2018-11-12:
Не воспроизводится. На твои примеры результаты у меня местами отличаются, в том числе на последний тест:

* `uefiextract bios.bin 91B4D9C1-141C-4824-8D02-3C298E36EB3F -o IRST_f -m file` — код 0, папка с ffs
* `uefiextract bios.bin 91B4D9C1-141C-4824-8D02-3C298E36EB3A -o IRST_f -m file` — код 8, папки нет (**другой код**)
* `uefiextract bios.bin 91B4D9C1-141C-4824-8D02-3C298E36EB3F -o IRST_b -m body` — код 0, папка с body
* `uefiextract bios.bin 91B4D9C1-141C-4824-8D02-3C298E36EB3A -o IRST_b -m body` — код 8, папки нет (**другой код**)
* `uefiextract bios.bin 91B4D9C1-141C-4824-8D02-3C298E36EB3F -o IRST_b_t10 -m body -t 10` — код 0, папка с body (**выше опечатка, но я исправил**)
* `uefiextract bios.bin 91B4D9C1-141C-4824-8D02-3C298E36EB3A -o IRST_b_t10 -m body -t 18` — код 8, папки нет (**другой код**)
* `uefiextract bios.bin 91B4D9C1-141C-4824-8D02-3C298E36EB3F -o IRST_b_t18 -m body -t 18` — код 8, папки нет (**другой код**)

Код 8 — `U_ITEM_NOT_FOUND`, код 1 — `U_INVALID_PARAMETER`.

Также взял винду в виртуалке и попробовал твой последний пример, папка не создалась, но код 1, да. Видимо что-то кривое в винде:

```
PS C:\Desktop> .\UEFIExtract.exe .\bios.bin -o IRST_b_t18 -m body -t 18
parsePadFileBody: non-UEFI data found in pad-file
parsePadFileBody: non-UEFI data found in pad-file
findFitRecursive: real FIT table found at physical address FF2C0100h
---------------------------------------------------------------------------
     Address      |   Size    |  Ver  | CS  |          Type / Info
---------------------------------------------------------------------------
_FIT_             | 00000070h | 0100h | 00h | FIT Header |
00000000FFB80090h | 00007000h | 0100h | 00h | Microcode | LocalOffset: 00000000h, CPUID: 00050654h, Revision: 02000043h, Date: 26.01.2018
00000000FFB87090h | 00007400h | 0100h | 00h | Microcode | LocalOffset: 00000000h, CPUID: 00050652h, Revision: 80000037h, Date: 02.05.2017
00000000FFB8E490h | 00007800h | 0100h | 00h | Microcode | LocalOffset: 00000000h, CPUID: 00050651h, Revision: 8000002Bh, Date: 08.02.2016
00000000FFB95C90h | 00007800h | 0100h | 00h | Microcode | LocalOffset: 00000000h, CPUID: 00050650h, Revision: 8000002Bh, Date: 08.02.2016
00000000FFB9D490h | 00018000h | 0100h | 00h | Microcode | LocalOffset: 00000000h, CPUID: 000906E9h, Revision: 00000084h, Date: 21.01.2018
00000000FFBB5490h | 00017800h | 0100h | 00h | Microcode | LocalOffset: 00000000h, CPUID: 000506E8h, Revision: 00000034h, Date: 10.07.2016
PS C:\Desktop> echo $LASTEXITCODE
1
```

Наиболее вероятно, что какая-то проблема у тебя в скрипте, так что я пока закрываю эту проблему. Если сможешь найти шаги для воспроизведения, создавай новый багрепорт.

closed 2018-11-12 by vit9696

LSSoniX, 2018-11-12:
Странно.. С кодом ладно, но у меня, хоть тресни, пуста папка всё равно папка создается. Хоть в PS, хть в cmd пускай. Даже в виртуалке ХР//

vit9696, 2018-11-12:
Сейчас есть проблема с длиной пути на Windows вроде как, может, слишком длинный путь?

LSSoniX, 2018-11-12:
Да я уже до "C:\1\uefiextract bin бда-бла,". сократил и уже на буке проверил, таже картина.
ОСьки вин7 и ХП. Возможно, что на 8-10 такого и нет, но мне не проверить.

vit9696, 2018-11-12:
Хм, я проверял действительно на 10. У меня должны быть виртуалки с этими операционками, я домой приеду вечером и посмотрю.

LSSoniX, 2018-11-12:
Я вот что заметил, буквально только-только выдаются сообщения и окно начинает закрываться и буквально тут же создается папка. То есть получается на закрытии приложения.

vit9696, 2018-11-12:
Воспроизвёл в конце концов. Отправил фикс.

LSSoniX, 2018-11-12:
?? 10-ка же улаляла..
Забрал А53. Спасибо. :)

PS Если не трудно, то компильни пожалуйста HexFind и FindVer,  они в UBU-Helper находятся.. Я только под винду собрать могу. Спасибо.

LSSoniX, 2018-11-12:
@vit9696 
Запутали всё. Этот топмк был с предложениями, а проблемы решали в #147 
Надо отсюда поудалять лишниие посты и переоткрыть заново на будущее..

vit9696, 2018-11-12:
На десятке есть подозрение, что не создавалось. Потому что на маке у меня тоже не создаётся и даже не заходит в этот код. Какая-то ерунда в логике, но сейчас не очень хочу изучать.

Поудалять — проблематично. Лучше новый баг открыть, и туда записать то, что осталось.

Утилиты собрал: [ubu-utils.zip](https://github.com/LongSoft/UEFITool/files/2573221/ubu-utils.zip)

LSSoniX, 2018-11-12:
Да и бог с ним, главное, что работает как надо.

Хорошо, предложения тогда заново пересоздам попозже в новом топике, там не только мой интерес.

За утильки спасибо. А название изначально не я придумал. ;)

LSSoniX, 2018-11-14:
@vit9696 
Приветствую!
Слушай, я косяк нашел. Сразу не посмотрел, а сейчас на питон начал переводить и обнаружил следующее.
У асусов и асроков есть Capsule Header по 2 и 4 кило.
Раньше при разделении
`UEFIExtract bios.bin 4A3CA68B-7723-48FB-803D-578CC1FEC44D`
Бивис https://cloud.mail.ru/public/64Lc/8vuiN6yVU
получали 3 файла header, body и info. Теперь body нема, только header и info

vit9696, 2018-11-14:
Записал, постараюсь глянуть на днях.

vit9696, 2018-11-14:
Должно быть исправлено.

LSSoniX, 2018-11-15:
Появился. Спасибо. :)

LSSoniX, 2018-11-17:
@vit9696 
Приветствую!
Отмени сборку t20181114
Это мой косяк в коде был. Использовал os.replace, а он переносил файл. Сейчас исправил на shutil.copyfile и всё пучком, те body.bin на месте был, а я тормознул.
Извини за ложный баг. :)

LSSoniX, 2018-11-17:
@vit9696 
Приветствую.
Извини, совсем запутался в этих сборках.
Перепроверил еще оаз.В общем текущая - рабочая, а предыдущая нет.
Оставляем как есть. Извни еще раз. :)

vit9696, 2018-11-17:
Всё хорошо, едем.

LSSoniX, 2018-11-17:
Хорошо, спасибо. Сегодня-завтра в пакете выложу, пока жду обноления МСЕ, да в питоне копаюсь.

LSSoniX, 2018-11-18:
@vit9696 
Доброго здравия.
Маленькая мелочь, не существенная, но приятная при тестировании
Добавить к отображению версии у всех утилит build, к примеру
URFIExtract/Find/Tool Axx build xxxxxxxx.x
xx - основной релиз
хххххххх,х в виде даты 2018117 и после точки цифру очередной тестовой сборки.
или что то в этом роде. Проще ориентироваться при тестах будет.

И еще вопрос с оверов, по послдним релизам
> при первом запуске лезет проверять доступные smb шары
> Это нормально?
>

vit9696, 2018-11-18:
Привет,
По поводу версии билда, сделай отдельно issue. Однако мне казалось, что при запуске без параметров версию должно печатать.

Про SMB шары ерунда какая-то. Никто туда сознательно не лез, так что либо случайное совпадение, либо винда сошла с ума. Чисто теоретически могу предположить, что какая-нибудь винапи функция, которую мы раньше не юзали, зачем-то подгружает эту информацию.

LSSoniX, 2018-11-18:
@vit9696 
Версии отображаются, тут всё нормально, просто когда начинаем очередные сборки тестить, то путаница иногда возникает какой файл был какой есть и тп.
Как я сказал это не критично. При следующмх тестах будц просто по пкпкам раскилывать, так оно проще будет.

Кстати плодёж файлов не совсем убрался. Вот выловил сегодня.
https://cloud.mail.ru/public/64Lc/8vuiN6yVU
`uefiextract bios.bin DE3E049C-A218-4891-8658-5FC0FA84C788 -o test - m file`
в бивисе с такмс GUID файл 1, но на выходе 3 FFS, как обычно по количеству секций