LongSoft/UEFITool #146: UEFIReplace: "No replacements can be applied to input file"

rcpao, 2018-09-29:
/home/u1/Downloads/UEFIReplace_0.25.1_linux_x86_64/UEFIReplace OVMF_CODE-EnTier-18873.fd 1214E36F-68E1-4A6B-9F6C-D9CD0CE59C74 7 EnTier.ffs -o OVMF_CODE-EnTier.fd -asis
**No replacements can be applied to input file**

UEFI Image > FFSv2 > Volume image > GUID defined > Volume image > right after PartitionDxe is my DXE driver with GUID 1214E36F-68E1-4A6B-9F6C-D9CD0CE59C74 which I want to replace with a new EnTier.ffs file.

Do I have the arguments wrong or any hints what I am doing wrong?  I can do this manually using UEFITool 0.25.1, but I want to do this from bash script or batch file as part of my build process.

vit9696, 2018-09-30:
Is this GUID FileGuid? To be honest, should work from what I remember. Mind passing me the file for testing? Will try to find some time to check it out.

rcpao, 2018-10-01:
I believe my input file OVMF_CODE-EnTier-18873.fd was corrupt.  Replacing it with a fresh copy resulted in "File replaced".  Sorry for the false issue report.

closed 2018-10-01 by rcpao