LongSoft/UEFITool #153: Changes to IntelGopVbt brick machine

gnidorah, 2018-12-09:
If I edit LegacyVideoRom in UEFITool everything is fine, but If I edit IntelGopVbt machine bricks.
For editing I'm using Intel BMP program (I'm using the same bsf script I use to edit LegacyVideoRom if that matter).
Also for some reason there are 4 matches for "vbt skylake" string in LegacyVideoRom and 6 matches in IntelGopVbt. I believe there are should be only one for each.
To edit I've used v0.21.5 instead of latest v0.25.1. Could it be the culprit?

vit9696, 2018-12-10:
I am not too familiar with the videorom patching, but have you tried the latest UEFITool? Also, I believe there is nothing strange in gop and legacy roms being different.

gnidorah, 2018-12-16:
That helped thanks, I've also worked more selectively this time.

Here are the new steps and modified bios (minimum brightness changed from 15 to 0 and PWM is changed from 220hz to 2000hz):
1. Opened firmware in **UEFITool v0.25.1**
2. Figured out that relevant sections are only DDE5BDB6-07F3-4CAD-A40C-52BAC366717D for LegacyVideoRom and CEDBA4B5-2184-496E-8764-D4BFB8FD0617 for IntelGopVbt
3. Used 1050 OROM script to edit both sections in **IBMP**
4. Adjusted only default mode (mode 3)

[bios.zip](https://github.com/LongSoft/UEFITool/files/2683417/bios.zip)

closed 2018-12-16 by gnidorah