LongSoft/UEFITool #154: UEFIReplace: reconstructVolume: root volume can't be grown

KingBonecrusher, 2018-12-23:
I failed to remove a dxe driver with latest UEFITool 0.25.1. I have also tried uefireplace cli tool, but same error. I have extracted the volume -->

https://www.file-upload.net/download-13443293/Volume_FFSv2_EfiFirmwareFileSystem2Guid.vol.html

KingBonecrusher, 2018-12-23:
Update:

Removing at least 4 or 5 dxe modules at once with UEFITool 0.25.1 is working for me. It seems the compression of uefitool is slightly bigger as the original from the dump. Is there any way/option to make a stronger compression?

vit9696, 2018-12-24:
The issue most likely happens due to hardcoded LZMA_DICTIONARY_SIZE value (0x800000), which is most likely bigger in your files. The dictionary size in modern BIOSes is present as the first UINT32 of the data blob, and it needs to be handled (thus the the TODO in LzmaCompress.c).

I do not foresee this getting fixed in master in the nearest future, and NE still has a long way to builder introduction, so what you could temporarily try is using your own value and recompiling UEFITool with it.

KingBonecrusher, 2018-12-24:
Thanks, your hint has done it! Can be closed...

vit9696, 2018-12-24:
Glad to hear. It remains to be an issue though, I will leave it open until we handle compression buffer size properly.

commit 4bee991c (2018-12-30): Implement custom LZMA dictionary size support, fixes #154
closed 2018-12-30 by vit9696