LongSoft/UEFITool #156: Support fixed address files in builder

vit9696, 2019-01-03:
APTIO firmwares have an extension called ROM hole, which load at fixed address and should not change it after a rebuild. Reference GUIDS:
```
0x05ca01fc, 0x0fc1, 0x11dc, 0x90, 0x11, 0x00, 0x17, 0x31, 0x53, 0xeb, 0xa8
0x05ca01fd, 0x0fc1, 0x11dc, 0x90, 0x11, 0x00, 0x17, 0x31, 0x53, 0xeb, 0xa8
```

skochinsky, 2019-01-04:
Another reference:  https://github.com/syscl/ASUS-H67-series/blob/master/BIOS_Utilities/Change%20Logo%204.52.0.37/ChangelogoReadMe.txt

> 7. Logos in ROM hole files.
> 
> 	Hole files have fixed locations and lengths within the ROM image. They are preceded by a protective Firmware File System header which must be of the RAW type. Sixteen firmware file GUID names are reserved for this purpose. ChangeLogo can replace or extract the data from these files. The program will inspect the raw data and attempt to identify the logo type. 
> 
> Sixteen GUIDs are defined for ROM hole logos:
> 	hole0   05ca01fc-0fc1-11dc-9011-00173153eba8	
> 	Hole1   05ca01fd-0fc1-11dc-9011-00173153eba8
> 	hole2   05ca01fe-0fc1-11dc-9011-00173153eba8
> 	hole3   05ca01ff-0fc1-11dc-9011-00173153eba8
> 	hole4   05ca0200-0fc1-11dc-9011-00173153eba8
> 	hole5   05ca0201-0fc1-11dc-9011-00173153eba8
> 	hole6   05ca0202-0fc1-11dc-9011-00173153eba8
> 	hole7   05ca0203-0fc1-11dc-9011-00173153eba8
> 	hole8   05ca0204-0fc1-11dc-9011-00173153eba8
> 	hole9   05ca0205-0fc1-11dc-9011-00173153eba8
> 	hole10  05ca0206-0fc1-11dc-9011-00173153eba8
> 	hole11  05ca0207-0fc1-11dc-9011-00173153eba8
> 	hole12  05ca0208-0fc1-11dc-9011-00173153eba8
> 	hole13  05ca0209-0fc1-11dc-9011-00173153eba8
> 	hole14  05ca020a-0fc1-11dc-9011-00173153eba8
> 	hole15  05ca020b-0fc1-11dc-9011-00173153eba8
>

slavonnet, 2021-01-26:
confirm last NE [alpha[]
[X9DRTiB5_407.zip](https://github.com/LongSoft/UEFITool/files/5876112/X9DRTiB5_407.zip)

NikolajSchlej, 2022-09-13:
Fixed items are supported in NE for a while already, AMI ROM holes support will be in the next release.

NikolajSchlej, 2022-09-15:
Fixed by #294.

closed 2022-09-15 by NikolajSchlej