LongSoft/UEFITool #157: Broken dump breaks UEFITool (old engine)

KingBonecrusher, 2019-01-07:
This dump breaks **UEFITool 0.25.1** and makes the whole system unstable. **UEFITool NE** shows instead *Image Truncated* and  *parseIntelImage: IE region is located outside of the opened image. If your system uses dual-chip storage, please append another part to the opened image*

I know this dump is defect, and **UEFITool NE** cannot process, but the old engine brings the whole system into unstable mode, so a error message is as better result...

https://www.file-upload.net/download-13461581/cracked_dump.rom.html

vit9696, 2019-01-10:
There are quite many ways to break UEFITool (both master and NE) with specifically crafted data. If security is a concern, I would suggest running it in a sandboxed environment.

While I do not foresee any further fixes being backported to master, I will leave this issue as a reminder to improve handling of untrusted data.

I believe a good start will be adding UEFITool new engine to Google ossfuzz project:
https://github.com/google/oss-fuzz

NikolajSchlej, 2022-09-13:
I'm mass-closing all issues that are specific to the old engine, because no further work is planned to be done on it, and all the bugs that are present there are to remain there forever. The original engine had a good run, but it needs to go in peace.

closed 2022-09-13 by NikolajSchlej