LongSoft/UEFITool #158: UEFiTool NE & UEFIFind NE - проблемы с поиском

LSSoniX, 2019-01-10:
Всем доброго здравия!

Только сейчас обратил внимание, что не работает совмещенный поиск header+body и похоже давно.
К примеру ищем такую строку
'728508177F37EF448F4EB09FFF46A070..AA01........F801'
Это заголовок микода и часть тела.
В результате
[image]
UEFIFind тоже ничего не находит
'uefind bios.bin all count 728508177F37EF448F4EB09FFF46A070..AA01........F801'
Но если убрать в конце 01, то всё ОК.
Также UEFIFind по строке выше без концовки 01, вместо 3-ёх найденных показывает, что нашел одно совпадение, что не есть хорошо, тк аргумент count считает количесство совпадений.

Со старым УТ 0.26.0 таких проблем нет
[image]
Файл для тестов https://cloud.mail.ru/public/FPwS/6WhQirkjN

LSSoniX, 2019-01-24:
@vit9696 
Доброго здравия. Да ничего страшного, время не имеет значения, "главное чтобы костюмчик сидел".
Спасибо.

PS Кстити, хотел попросить, есть возможность прикрутить возможность подставлять '.' вместо символа в Hexfind (Findhex) как в UEFIFind? А то в обычных файлах тоже не всегда точный паттерн задать можно.

vit9696, 2019-01-24:
Может, лучше использовать [ack](https://beyondgrep.com/install/) какой-нибудь? Там полноценные регулярки и ещё много чего. У меня вот так сделано:

```
function ackw() {
  if [ "$#" -ne 2 ]; then
    echo 'Usage: ackw <pattern> <dir>'
    return -1
  fi
  ack -s --type-set='all:match:.*' -k "$1" "$2" --output=ackwline | sed 's/:[0-9]*:ackwline$//' | uniq
}
```

LSSoniX, 2019-01-26:
Для линухов то да, но для форточек увы..
Лады, не обращай внимания.