LongSoft/UEFITool #162: Unable to save modified 1603 ASUS Prime X299 Deluxe firmware

KGP, 2019-02-01:
Hi,

there are apparently issues in saving modified ASUS X299 BIOS firmware recently, which is at least the case with BIOS firmware 1603 for the ASUS Prime X299 Deluxe and BIOS Firmware 0404 for the ASUS Prime X299 Deluxe II. 

Error: Invalid UEFI Volume! 

Could you please fix this issue as soon as possible within some minor UEFI Tool update?

Many thanks in advance,

KGP

[image]

vit9696, 2019-02-01:
This most likely is a known issue. There are several related issues to that in the issue tracker.

UEFITool 0.2x.0 is end of life. UEFITool NE builder is not ready yet. Your best approach is to temporarily use MMTool and wait for UEFITool NE to land.

KGP, 2019-02-01:
It is not really a known issue. Saving the ASUS Prime X299 Deluxe firmware after modification under UEFITool worked absolutely flawless up to BIOS firmware 1503. The recent error is totally new and only appears with BIOS firmware 1603 or any firmware version provided for the ASUS Prime X299 Deluxe II.

Anyway , from your reply above I fully understand that UEFITool 0.2x.0 is end of life and any respective update might not be possible anymore. 

Many thanks for your reply.

vit9696, 2019-02-01:
Actually, the issue here is unrelated to UEFITool bugs. I checked the screenshot, and it shows that the the driver you added simply does not fit the volume.

closed 2019-02-01 by vit9696

mrmggo, 2019-03-03:
0.25.1
Can modify the 1603 and 1704 versions of bios
Note that you cannot inherit the settings backup of the previous bios version.

KGP, 2019-03-03:
One just has to reduce the splash screen image for its implementing in BIOS firmware versions > 1503. The last comment of vit9696 above was totally correct and adequate. Issue solved!

All the above is not about BIOS settings and it is commonly known that saved BIOS settings or only valid within the same BIOS version and cannot be transfered or restored within other BIOS versions.