LongSoft/UEFITool #165: extent for report

Raven19409, 2019-02-14:
предлагаю добавить (в конец отчета) интегральные характеристики типа:
 +---------------------------+
 | Report included:              |
 +---------------------------+
 | Intel Flash Image    =    1 |
 | Descriptor Region  =    1 |
 | GBE Region            =    1 |
 | ME/TXE Region      =    1 |
 | BIOS region           =    1 |
 | Volume(s)              =   11 |
 | Pad Block(s)           =    2 |
 | File(s)                     =  627 |
 | Vol Free Block(s)    =    8 |
 | Section(s)             = 2175 |
 +---------------------------+

vit9696, 2019-02-15:
Привет, дело хорошее, хотя мне сложно представить use-case для подобной информации. Наличие количества свободных блоков я ещё понять могу, скажем для какой-нибудь модификации в далёком будущем, но остальное относительно. По идее тут работы немного, если пришлёшь патчик, я его солью.

Raven19409, 2019-02-18:
Иногда приходится сравнивать 2 различных прошивочки...
Иногда приходится сравнивать 2 различных модификации одной и той же прошивочки...
Иногда приходится просматривать отчеты, созданные различными программами/скриптами.
Иногда они сильно отличаются, а понять структуру полезно...
А вот насчет патчика вряд ли получится. Занимаюсь абсолютно другими вещами...
А почему предлагаю, из опыта знаю, часто не хватает!
Лично я-то такие вещи у себя формирую и мне этого достаточно, просто предложение расширить функционал... На практике проверено.
Ну а если никому не надо или сваять некому, то Аминь!
Вообще на практике нужно больше и разных отчетов... (при серьёзной работе).

NikolajSchlej, 2019-07-24:
Совершенно непонятно, зачем это все нужно в основной программе, потому что отчеты лучше всего сравнивать не по количеству элементов, а напрямую. На данный момент это не нужно, кому хочется больше отчетов - код открыт, пулл-реквесты принимаются.

closed 2019-07-24 by NikolajSchlej