LongSoft/UEFITool #167: UEFIExttact - пара вопросов

LSSoniX, 2019-02-28:
@vit9696 

Доброго здравия.
Хочу уточнить пару моментов.
1. Параметр строки
`uefiextract BIOS_file GUID`  с <-o> и без
должна быть распаковка содеhжимого GUIID?  Если не задавать метод и секция, то ошибка 35 в любом случае.
2. Для аргументе <-m file>, если FFS в бивисе сжат, то извлечь этот FFS в расжатом виде. Возможно что то придумать? Суть в том, что когда в FFS куча файлов, то просматривать их по отдельности утомительно, проще глянуть в один файл, найти что нужно для последующих действий.

Спасибо.

vit9696, 2019-03-08:
Привет,

Я почему-то пропустил этот issue и увидел только соседний. Извини за задержку.

1. По идее это баг, хотя я не уверен, как правильно решить относительно создания директорий. Сейчас она создаёт все директории по ходу, но сохраняет файл только с нужным GUID.

2. В NE нет интерфейса для этого пока, сделать можно и нужно (добавить ключик unpack или что-то подобное), но это чуть дольше 15 минут. Оставлю issue открытым и буду иметь в виду.

LSSoniX, 2019-03-08:
@vit9696 

По первму я чисто случайно столкнулся. Искал метод соединения извлеченных файлов в один. Решение я нашел, а вопрос остался.
По сути он (баг) не критичен, тк когда не указан метод <-m>, то не понятно, что конкретно должно быть извлечено файл целиком или разделанный как шпрот. :)).

По второму вопросу думаю, что пока закроем. Тк решать лучше будем в комплексе отсюда #150

vit9696, 2019-03-08:
Действительно, лучше тот issue иметь в виду. Закрываю.

closed 2019-03-08 by vit9696