LongSoft/UEFITool #168: UEFIPatch EVGA X299 bios issue

antonsemanyk, 2019-03-05:
I'm trying to patch EVGA X299 bios, and it gives me next output:

`
parseFile: non-empty pad-file contents will be destroyed after volume modifications
patch: replaced 8 bytes at offset 4389h 81E10080000033C1 -> 9090909090909090
patch: replaced 7 bytes at offset 2B0Ch 742CB9E2000000 -> 752CB9E2000000
patch: replaced 8 bytes at offset 1510h BE0080000023CE0B -> BE0000000023CE0B
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
Image patched
`

Does this mean that patched bios will not work?

vit9696, 2019-03-05:
It means that patched BIOS *may not* work. If you do not have a SPI flasher, then it indeed is better not to update your BIOS with the resulting file.

Closing as duplicate of https://github.com/LongSoft/UEFITool/issues/156, which should also workaround this issue.

closed 2019-03-05 by vit9696

psyborg55, 2020-08-12:
hey @vit9696 

i am running into this same issue, and file flashed is broken, e.g. laptop does not boot.

can you tell me, or point into the sources how the checksum of a modified PEI module is calculated?

i was suggested by someone else to use uefitool as it should fix checksum, but it modifies a lot more bits and the whole process ends up with unusable bios file.

i am trying to change two bits inside a module, thus no module size change and i think no rebasing or addresses update are needed, only a checksum fix to see whether changed bits make any difference on machine.