LongSoft/UEFITool #169: Can not rebuild/insert/replace/remove EFI files in FV,they are all grey

comicsheep, 2019-03-14:
**Parser log**
```
parseIntelImage: SPI flash with unknown JEDEC ID 9D7F7A found in VSCC table
parseIntelImage: SPI flash with unknown JEDEC ID EF4F18 found in VSCC table
parseMeRegion: ME region is empty
parseVendorHashFile: new AMI hash file found
parsePadFileBody: non-UEFI data found in pad-file
findFitRecursive: FIT table candidate found, but not referenced from the last VTF
findFitRecursive: FIT table candidate found, but not referenced from the last VTF
findFitRecursive: real FIT table found at physical address FFBB0000h

```
**security log**
`New AMI hash file found at base 1D90A28h
Protected ranges:
Address: FF120000h Size: 39978Ch
Hash: 34D97D3A57098B4EB3CC953726AF4B868E1CB543A534CF5907615634675538DD
Address: FF4BA78Ch Size: 20B474h
Hash: F869F2B7456CAEBA8AC279B1611633459C55961E73B019B494A7B88481C65203
------------------------------------------------------------------------

S-ACM found at base 1BC0000h
ModuleType: 0002h         ModuleSubtype: 0001h     HeaderLength: 00040000h
HeaderVersion: 00000000h  ChipsetId:  B007h        Flags: 0000h
ModuleVendor: 8086h       Date: 18.04.2018         ModuleSize: 00040000h
EntryPoint: 0001362Eh     AcmSvn: 0002h            Unknown1: 00000000h
Unknown2: 00000000h       GdtBase: 00001B5Ch       GdtMax: 00000020h
SegSel: 00000008h         KeySize: 00000100h       Unknown3: 0000023Ch

ACM RSA Public Key (Exponent: 11h):
A108074888E288BEA4DCAE0D7920188BA3DDD5349EA713DE75C7426CC100625C
D2AD42D8720C5322C61FE341A2C32476B444C61367C24AF9B347117463315D64
4D991FEF2BC34B1A5A36A49C1BEF84211F07E8428CE60B0D2861089BA2947119
6EF7561A1F6C9513C3EBF7DC0ED1D1C7FBE56DFB438C4C317778116899B866AB
F9B653010F58DD030A7086A17F0002471D096CFCDE8C50B0BE05F2BAE41A9871
C614F31F7D73655D765B6F3E117840A172F7C8608377F3C4F23B797936B53825
AD6B7B8BC0FF96686A2CA237692B41D8686BBA1DC110909A6B172E831F71958A
2FCB015B16449F511FD557E60B9435EBCAA6D1085F63186718334779C2FA1999

ACM RSA Signature:
370E3E7F3D4015CC9D36500C61E20F2E55F1CCFDCF17A235D3EE0351673706D7
339CFA18AE9ECCE1246F22685D14DFECCB511DFE9E00169FCDA16FFC3A607FAD
2E2617B821CE8DCA2E78D4D02CD3D75BDEF4E82D93841C4E7C2FBCD34E2D05DA
AB350770D3523C66C6CB16176535EF6BFAD48936283727CF130C66B936AA4E5A
4C3963B1B2C6FB88DE417EDF7441DE956675003846BE10264B7969ED0875523E
E721CD16A5398E8DD8D6A906E470467395929A8937B92E1AB78A727AD8819DF9
3E984EDDA228B1DFD8F751D74F663C13868C3BEBE8BE1117492135F3104523D7
31577803D79B7BDE4E2C7E5C5202301E2E594DDEA7490EF9562D9F82240C0F31
------------------------------------------------------------------------

Intel BootGuard Key manifest found at base 1DB9080h
Tag: __KEYM__ Version: 10h KmVersion: 10h KmSvn: 00h KmId: 01h

Key Manifest RSA Public Key Hash:
BFFD53DC8ED542D9E8EF4E322B1AB901B00AB7A28C707AE5C6376870BF869613

Boot Policy RSA Public Key Hash:
61E8E077ACDE2D81F2B1BBC6305ACF7630546E8BB4CF62F03FBB925E2473B3E7

Key Manifest RSA Public Key (Exponent: 10001h):
CD092356C876350ADB8B27C4B683798CFBEE243289A2F64EA08345E599A33FDA
7DF8D810164439112BCC8C1AB8FB704C57E55AC8E256BD58F16238E2B29E16BB
5E09EA3E7BD6EE69748EBCD2CB12F823EDCB359D43DA8BA1AD1E215A4A24E6FB
7C0792A3200C0397F424E8F052E436B41E5F737C952C19C61A07AD7AFD4A16F8
CBB11283F6B7171FBE68BD4750A9A1BF012BB9ACD9555B8EBD624C2E34D9D9E6
6BA15A1F6E76133FC9D65B6FE38027483F08A587490A4CF0D41B5000561A8396
C82869EBC06D73FBB6C05C0B7FF2B90ABDC21B793C479891CDF61C45D2F7DBCA
90741169C9C24A90BC5F840E9A916745EDAAABA2D2D68695E5A4E09344D4B9DE

Key Manifest RSA Signature:
C9DF9593A25B6B15CCD4D5A422AECC6DDA9620CE0EBF75C76F266181639CBED3
4BE21D0584D6B4ECE4D981D97B4ADC7B461DD5C4D9E86EE96DB21768CE755D75
42F4EF309455548E1C82BDA8299E8E3D0BB49F728A97DD0996F989972C7A1574
930A5811CF256B4F65474BBF3618D0AC14BCFEF9B87D23D499ACBAB92E25A6E3
69F570EA39BCFEA35DE3CDB27612A4CD8DCF9BFDE515DDA8774BF8EC9ADB3B90
C3834E5DA05E2EBAC6F3469EFCC0BB81DBC364F731D12BF8BC74E3DEA31AE4EA
02071A4F1F3D033298A5EB0601EEBC4FA3CACBA914B26707D10B421FAC296D38
5B8E3D292B72F3273D2207378C2B5BB2A8B9CEDD446549E6831E984900EE224E
------------------------------------------------------------------------

Intel BootGuard Boot Policy Manifest found at base 1DB8000h
Tag: __ACBP__ Version: 10h HeaderVersion: 01h
PMBPMVersion: 10h PBSVN: 00h ACMSVN: 02h NEMDataStack: 0180h

Initial Boot Block Element found at base 1DB8028h
Tag: __IBBS__       Version: 10h         Unknown: 0Fh
Flags: 00000006h    IbbMchBar: 00000000h VtdBar: 00000000h
PmrlBase: 00000000h PmrlLimit: 00000000h  EntryPoint: 00100000h

Post IBB Hash:
0000000000000000000000000000000000000000000000000000000000000000

IBB Digest:
66DA56081E48A50279C113E969FD7FF6850299225516BB110673DD5BD1616B76

IBB Segments:
Flags: 0000h Address: FFC00000h Size: 0019A380h
Flags: 0000h Address: FFDB7FC0h Size: 00000040h
Flags: 0000h Address: FFDB9000h Size: 00000080h
Flags: 0000h Address: FFDBA080h Size: 00245F80h

Boot Policy Signature Element found at base 1DB80DDh
Tag: __PMSG__ Version: 10h

Boot Policy RSA Public Key (Exponent: 10001h):
695E54C4FB71DC9C7C72699F719B1AFB5D409D3CE1B445E1DC4AA0741E62C635
9D4549D03A3AEB0980E886D7B2D029ACDE84918974A1224C6C1DFB08210A5E67
3513C0A0C6C507C0D9E7177DF6E3BBCD7340EFF31EE12514202B2E9FF921D65D
4157D6CD6327C15C9B4A2060A50A3114B4E01E87DC4959C0C242064B5C06FFA8
3B6AC081B31CF5AA754DEC39EDB26E8595BCBCF1EF6897D70583642E59257F11
B9CD6D95C14E81B22783AED2BBDCA3179470BD67304683A10C335466C629F2EA
2DC22C68785782572FA445A6BED8617DC59447D95FD38BBABAF7B5B1D672E0E0
90DD96D09552BB6D246714D78E64DF4754C7C64CFADF0A14DDA7F32667903EC8

Boot Policy RSA Public Key Hash:
61E8E077ACDE2D81F2B1BBC6305ACF7630546E8BB4CF62F03FBB925E2473B3E7

Boot Policy RSA Signature:
2E74EB0CBC95BE09526C94BD0D28790BCFFE22F91BBFC1F5553A04CC049403E7
9B3E6C593FF6569CE9336F21776044772C29B1A7D88F197E867DCA9E0BB56655
C6AD43FDC876E56A5818DD4BEF227F9FDF65B68C5B3F821BD772855C87B5C445
4062E4E6F5CACEE5E671237D9826EA7392C81B6F7681608F0190E07231CB2BE0
648C744611886F7D37D2CC7012405164276130FF57D212A3D3E99DB62EA23614
78416A663F025A7148FDD034450FECEB57BA29855F05A1C748AC7323E8487AD1
C5C22FE4E9381C178B3A828359925CC04BF82DCFA559BAE9248F66CB9BE17C06
E9FD28A1546D2504DF9145C8AE12D54F29AEFCEB0B7A57587B79E9B3737BC9BC
------------------------------------------------------------------------

`

comicsheep, 2019-03-14:
version:UEFITool_NE_A55_win32
System Information
        Manufacturer: Inspur
        Product Name: SA5212M5
        Version: 00001
Base Board Information
        Manufacturer: Inspur
        Product Name: NF5280M5
        Version: NF5280M5
BIOS Information
        Vendor: Inspur
        Version: 4.0.6
        Release Date: 03/07/2019

vit9696, 2019-03-14:
Duplicate of https://github.com/LongSoft/UEFITool/issues/67.

closed 2019-03-14 by vit9696