LongSoft/UEFITool #170: UEFITool / UEFIReplace v0.26.0 findFitRecursive: FIT table candidate found, but not referenced from the last VTF

LSSoniX, 2019-04-01:
Всем доброго здравия!

@vit9696 
Понимаю, что на старый движок уже почти не смотрим, но всёж..
Столкнулись вот с такой проблемой при замене FFS с микрокодомаи.
GUID 17088572-377F-44EF-8F4E-B09FFF46A070
Я проверил замену "как есть" и в УТ и в УР, везде один и тот же результат.
Сам FFS не в PEI ничго такого, но после замены когда смотрим результат замены в UEFITool NE, то видим следующее
соответственно прошив такой мод - кирпич.
Файлы бивиса и FFS
https://cloud.mail.ru/public/FPwS/6WhQirkjN

LSSoniX, 2019-04-01:
Упс. Почему то скрин не цепляется, загрузил к файлам.

Спасибо.

vit9696, 2019-04-01:
Привет,

Если коротко, то данная проблема не решаема в рамках старого движка: он не умеет фиксировать регионы на своих местах, и в итоге FIT оказывается в неверном месте.

Можно пробовать подставлять микрокоды одного и того же размера с оригинальными, тогда работать будет, но это костыль.

Закрываю как дубликат https://github.com/LongSoft/UEFITool/issues/156

closed 2019-04-01 by vit9696

LSSoniX, 2019-04-02:
Спасибо. :)
Мне уже пояснили, что там не так.