LongSoft/UEFITool #171: new NVAR FFS guid

Raven19409, 2019-04-11:
#define FV_BB_DEFAULTS_FFS_FILE_GUID \
    { 0xaf516361, 0xb4c5, 0x436e, { 0xa7, 0xe3, 0xa1, 0x49, 0xa3, 0x1b, 0x14, 0x61 } } 
example: "ASUS ROG MAXIMUS XI GENE"

vit9696, 2019-04-11:
Hello,

- who is the author of this GUID?
- where does the name come from?
- what is the format of the entry?

Thanks.

Raven19409, 2019-04-11:
1. Скорей всего AMI.
2. Скорей всего Aptio V (5.13)
3. Стандартный NVAR (как в {CEF5B9A3-476D-497F-9FDC-E98143E0422C}
 кусочек из UEFITool NE (последней) report:
 File        | Freeform  | 00DCDD00 | 00002D57 | F10DF525 | -- AF516361-B4C5-436E-A7E3-A149A31B1461
 Section  | Raw          | 00DCDD18 | 00002D3F | C8AF2F85 | --- Raw section

skochinsky, 2019-04-11:
reference: https://github.com/Durantlin/RAIDOOBMODULE/blob/c7091606b434a9cd0ca4cfab038d1605aa18339a/AmiModulePkg/NVRAM/NvramPei.c#L414

Raven19409, 2019-04-11:
Спасибо, Игорь!
Постеснялся светить большой кусок Aptio V!
Хотя ты прав! Источник этот ( в том числе)!

vit9696, 2019-04-11:
А, ну этот позор супермикры у меня есть. Посмотрю по остальным сорцам, когда оно появилось, и добавлю.

closed 2019-04-11 by vit9696