LongSoft/UEFITool #172: UefiPatch fails on this bios

chriz74x, 2019-04-12:
UEFIPatch.exe BIOS-Backup.rom
parseVolume: unknown file system FFF12B8D-7696-4C8B-A985-2747075B4F50
parseFile: non-empty pad-file contents will be destroyed after volume modifications
parseFile: non-empty pad-file contents will be destroyed after volume modifications
No patches can be applied to input file

File is here: https://www.dropbox.com/s/d3lt80b2j6y20bp/BIOS-Backup.rar

vit9696, 2019-04-13:
The warnings you see are fairly harmless (NVRAM data storage parsing only exists in UEFITool NE, and non-empty pad contents is something you have to live with). As there is actually no patch data in this BIOS, UEFIPatch works as defined by claiming that there is nothing to patch.

closed 2019-04-13 by vit9696

chriz74x, 2019-04-13:
Ok so there is nothing to patch but the bit 15 of 0xE2 registry is set to 1 so what can I do to unlock this bios ? It’s locked and there is no option in the menu to unlock it (cfg lock)

succeedmr123, 2022-02-03:
#metoo (slight pun intended)

Loical questions:

This volume is what? Which function does it make within UEFI BIOS file?

Can it be known what structure it is and then have a legitimate known description within the tool.

As its UEFI (BIOS), what comes to mind are problems like corrupting the structure, or tany tool inability to  calculate a correct checksum  after any modification or swap of the modules/volumes. which would probably defeat the purpose of owning your own machine and modifying the hardware functions to your own use case - many machines do verify UEFI checksum and then the system freeze for invalid chacksum (examples: e.g. HP laptops of newer build are notoriouzs for not allowing their users fully own the hardware they paid for by severely crippling the available options, and unverified UEFI flashing usually produces a soft brick)

Best