LongSoft/UEFITool #173: Can't patch intel DX79TO Bios

ColdfireMC, 2019-04-21:
I don't know if is planned to be supported, but I have an DX79TO motherboard with intel bios, and UEFI tool seems to work partially Not all GUID's are properly identified. Apparently those bios are "Upgrade based", especially considering some funny issue I had time ago (that resulted in a returned motherboard), when a buggy bootloader deleted one of the GUID's related to hard disk entries.

I want to modify bios to add nvme support adding the standard nvme ffs. If its needed, the serial port (actually IrdA) and the annoying intel startup bitmap, can be wiped off to gain the 5kb needed.

[SI0650P.BIO.report.txt](https://github.com/LongSoft/UEFITool/files/3100741/SI0650P.BIO.report.txt)

thanks for your time reading this.

NikolajSchlej, 2019-07-24:
You can't modify Intel's BIO files because they are covered by RSA signature that will become invalid after any modification.

closed 2019-07-24 by NikolajSchlej

McMacaron, 2021-07-26:
So they discontinued support for that motherboard. You can not even download old drivers or bios. Real weird attitude.