LongSoft/UEFITool #174: UEFITool/UEFIReplace v0.25.1/0.26.0 - lost dsta

LSSoniX, 2019-04-21:
Всес доброго здравия.

После внесения изменений в #137 в последствии обнаружились проблемы на некоторых биосах. Потеря сегмента кода в смещении 0xFFC000.
В основном это часто встречается на последних БИОСах мат плат Асус и MSI, возможно и у других брендов.
Суть в том, что любые изменения в различных томах, например в DXE или томе с резмещенными в нем микрокодами и/или _FIT_, затрагивают содержимое PEI тома, те как раз потерю сегмента кода в указанном выше смещении.
Версии УТ и УР 0,25,0 такой проблемы не создают.
Вот пример, были заменены FFS с миерокодами в версиях 0.25,0 (025_bios.bin) и 0.25.1 (0251_bios.bin). Приложил скриншоты.
https://cloud.mail.ru/public/2MwZ/8tuxhYFR7
Спасибо.

vit9696, 2019-04-23:
Привет,

После тех изменений стал теряться startup data array в конце тома (https://github.com/tianocore/edk2/blob/master/BaseTools/Source/C/GenFv/GenFvInternalLib.c#L104). Я не уверен, что мы это будем чинить в мастере, но спасибо за репорт.

dsanke, 2019-12-06:
When using UEFIPatch patching TE Image in the last volume, such as SiInitPreMem , it will cause the same problem with warning "rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working".

LSSoniX, 2020-02-15:
Всем доброго здравия!
Ребят, я вот, что подуал. А может прикрутить к УТ/УР/УП 0.25.1 и выше этакую прцедуру "костыль"?
Примерно так:
Глоьально описываем константу с теряющим PAD
`FFFFFFFFEAD0FF00F0000000000000000000272DFFFFFFFF`
и 2 переменных. Одна переменная будет отвчеать за статус наличия этого PAD, а во второй будет храниться оффсет размещения.

При открытии файла БИОС утилиты проверяют наличие сия PAD м если он еть, то соответствующие перменные получают нужные данные.
После сохранения изменений смотрим в переменную наличия сия PAD и если он есть, то проверяем есть ли по заранее известному оффсету данные. Если есть то всё ОК, а если нет то прописываем данные из константы в этот оффсет.
Ну как то так.

NikolajSchlej, 2020-03-06:
Придумаем что-нибуть, если руки дойдут, спасибо за идеи.

NikolajSchlej, 2022-09-13:
I'm mass-closing all issues that are specific to the old engine, because no further work is planned to be done on it, and all the bugs that are present there are to remain there forever. The original engine had a good run, but it needs to go in peace.

closed 2022-09-13 by NikolajSchlej