LongSoft/UEFITool #176: Is checksum working ok?

Chinoforest, 2019-08-15:
Hello. I'm not sure if this is an issue or a missundestanding on how UEFITool works. But here it goes:

I have this BIOS (from a laptop Sony Vaio VGN-NS135E): https://mega.nz/#!XtkDQSTC!QqRSYR5Y_gkBpUM9kJN073PZgVXyO1pRXDxstpzxhqM

I updated the microcodes this way:
With UEFITool 0.26 I extracted the body of the microcodes section. Updated them using HxD, and then replace that part of the BIOS with UEFITool. I saved the new modded BIOS (rebuilding it) and the produced file was this:

https://mega.nz/#!Sp9xgKjC!NCKvPNDfE09uhYmt1XMzVafj0Tf1kwZaKQiHI8HjdWI

My doubt here is: the header checksum and the data checksum of the microcode section remained the same. They didn't change! So I tried to verify that with HxD but I can't figure out how these checksums are calculated (I've tried multiple CRC variants and I couldn't find the correct one). The only thing I can say is that UEFITool NE tells me that both the original and the modded BIOS have valid checksums (which are the same!).

I also tried changing some random byte in one microcode, and UEFITool NE also tells me "valid" checksum... so I have some serious doubts on how it's working. 

Could you please tell me what I'm doing wrong or if it's a bug of UEFITool?

PS: I also tried UBUTool which uses MMTool and it produced different checksums but it removed the padding from the microcode section, so I didn't really liked the result. Here is the link in case someone needs it to investigate: https://mega.nz/#!ipsFDSxK!2MBl0tBIGLrvKslw25uczrTlYeEmM1sY19i4wuX8Lfg

THANKS!!! AND GREAT WORK!!

hardboost, 2019-08-15:
Checksum is not calculating 'cause it is fixed checksum 0x5A.
https://github.com/LongSoft/UEFITool/blob/4bee991c949b458739ffa96b88dbc589192c7689/ffsengine.cpp#L1261-L1270
File's header doesn't have FFS_ATTRIB_CHECKSUM attribute and its parent Volume section has Revision: 1

Chinoforest, 2019-08-15:
Ohhh I see. So I can just hex edit those microcodes and everything else in the BIOS will remain the same and the checksums will stay the same too? 

This is something particular to my BIOS? The fixed checksum I mean.

skochinsky, 2019-08-15:
AFAIK fixed checksums are pretty common. It's pretty weak anyway, so when integrity checking is really necessary, OEMs often use CRC32 using custom GUID-defined sections.  As for microcode, it's signed by Intel so there's no real need for additional integrity checks on top of it.

Chinoforest, 2019-08-15:
I undestand. So if my BIOS have a custom GUID with a CRC32 I don't have a way to know. Right?

What happens if you flash a ROM with an incorrect CRC32? (I know this is a difficult question). I just wanted to update the microcodes of my old notebook and I was checking all the possibly error I could face before taking the jump. And one of the things I wanted to check was if those checksums (specially the data checksum of the that section) was calculated correctly. Now I see it's fixed.

vit9696, 2019-08-15:
I am afraid you confuse ROM signature and section CRC. Microcode is loaded very early through the assembly, so there is no sense to make any checksum on it. Closing the issue as not a bug.

closed 2019-08-15 by vit9696