LongSoft/UEFITool #184: [Enhancement Req][NE][FFSParser][parseRawArea] parse $VSS in padding aera

lzp729, 2019-09-06:
With AMD BIOS built with Insyde EDKII, a group of var stores is in the padding area immediately after FV NVRam, please enhanced the RawAreaPaser function.

NikolajSchlej, 2019-09-06:
Please attach a sample file.

lzp729, 2019-09-06:
[gh2.zip](https://github.com/LongSoft/UEFITool/files/3585295/gh2.zip)

uploaded, for some NDA reasons, I have removed most DXE and PEI drivers, but still enough to show the $vss in Padding
Fixed: Yes
Base: 63B000h
Address: FFE3B000h
Offset: 63B000h
Full size: 45000h (282624)

lzp729, 2019-09-06:
and also, please note 
--------------------------------------
Fixed: Yes
Base: 60B048h
Address: FFE0B048h
Offset: 48h
Full size: 15FB8h (90040)
--------------------------------------
which caused an error msg with new engine 
> findNextStore: VSS store candidate at offset 48h skipped, has invalid size FFFFFFFFh.
Insyde EDKII use the area as BIOS setting storage, FFFFFFFFh means it's empty, should not be an error.
--------------------------------------
for the error msg
> parseVolumeHeader: unaligned volume
I can see that with full BIOS binary just like the reduced one, yet have not look into that

NikolajSchlej, 2019-09-06:
>should not be an error.
It is not an error, but an informational status message that might or might not indicate an error. This is extremely useful in catching empty volumes that aren't supposed to be empty, and I'll leave it as is because of that.

NikolajSchlej, 2025-03-16:
Found another sample of VSS store in a Padding.

[image_initial_123456.bin.zip](https://github.com/user-attachments/files/19272607/image_initial_123456.bin.zip)

NikolajSchlej, 2025-04-14:
Tried doing this in a trivial way by adapting already existing VSS and VSS2 parsers added in A70, turns out those out-of-volume VSS stores are even more fucked than we expected. Will need some more effort to get this fixed.