LongSoft/UEFITool #187: Different (compact) Apple VSS format?

nkaretnikov, 2019-12-24:
Did Apple switch to a different (compact) VSS format?

I used chipsec on a Mac Mini 2018 running macOS Catalina to dump the firmware, and UEFITool fails to recognize the VSS section.

After looking at the section, I realized that the format is different.

I don't want to share the dump, but here's a template I wrote for 010 Editor:

```
LittleEndian();

struct VSSHdr {
  UINT32 Magic;
  Assert(Magic == 0x53535624, "Invalid VSS magic");
  UINT32 Unk1;  // 1 in the dump
  UINT16 Size;  // including hdr
  UINT16 Unk2;  // 2 in the dump
  UINT32 NumOfEntries;
};

struct EfiGuid {
  UINT32 Data1;
  UINT16 Data2;
  UINT16 Data3;
  UBYTE Data4[8];
};

// Values taken from UEFITool.
enum <uint32> VSSAttributes {
    ATTR_NV = 0x00000001,
    ATTR_BS = 0x00000002,
    ATTR_RS = 0x00000004,
    ATTR_TIME_AUTH_WRITE = 0x20,
};

struct Variable {
  UINT32 Attributes;
  switch (Attributes) {  // for this particular dump
    case ATTR_NV | ATTR_BS | ATTR_RS: break;
    case ATTR_NV | ATTR_BS: break;
    case ATTR_NV | ATTR_BS | ATTR_RS | ATTR_TIME_AUTH_WRITE: break;
    default: {
      Printf("Attributes: 0x%x\n", Attributes);
      Assert(false, "Unexpected variable attributes");
    }
  };
  UINT32 NameSize;
  UINT32 DataSize;
  struct EfiGuid VendorGuid; // end of hdr
  wchar_t Name[NameSize / sizeof(wchar_t)];
  BYTE Data[DataSize];
};

VSSHdr vss_hdr;

Variable vars[vss_hdr.NumOfEntries] <optimize=false>;
```

And a Python parser:

```Python
#!/usr/bin/env python3

import struct
import sys

FILE = sys.argv[1]

def unpack_uint64_be(bytes_):
    return struct.unpack(">Q", bytes_)[0]

def unpack_uint32_le(bytes_):
    return struct.unpack("<I", bytes_)[0]

def unpack_uint16_le(bytes_):
    return struct.unpack("<H", bytes_)[0]

def unpack_uint8(bytes_):
    return struct.unpack("<B", bytes_)[0]

with open(FILE, "rb") as f:
    found_magic = False
    vss_magic = f.read(4)
    while vss_magic:
        if vss_magic == b"$VSS":
            found_magic = True
            break
        vss_magic = f.read(4)

    assert found_magic

    vss_unk1           = unpack_uint32_le(f.read(4))
    vss_size           = unpack_uint16_le(f.read(2))
    vss_unk2           = unpack_uint16_le(f.read(2))
    vss_num_of_entries = unpack_uint32_le(f.read(4))

    print(
        "vss_hdr:"
        f"\nmagic:              {vss_magic}"
        f"\nunk1:               0x{vss_unk1:x}"
        f"\nsize:               0x{vss_size:x}"
        f"\nunk2:               0x{vss_unk2:x}"
        f"\nvss_num_of_entries: 0x{vss_num_of_entries:x}"
    )

    while True:
        var_attrs = unpack_uint32_le(f.read(4))
        if var_attrs == 0:
            break

        var_name_size  = unpack_uint32_le(f.read(4))
        var_data_size  = unpack_uint32_le(f.read(4))
        var_guid_data1 = unpack_uint32_le(f.read(4))
        var_guid_data2 = unpack_uint16_le(f.read(2))
        var_guid_data3 = unpack_uint16_le(f.read(2))
        var_guid_data4 = unpack_uint64_be(f.read(8))

        var_name = f.read(var_name_size).decode('utf-16')
        var_data = f.read(var_data_size)

        print(
            "\nvar_hdr:"
            f"\nvar_attrs:      0x{var_attrs:x}"
            f"\nvar_name_size:  0x{var_name_size:x}"
            f"\nvar_data_size:  0x{var_data_size:x}"
            "\nvar_guid:       "
            f"{var_guid_data1:x}"
            f"-{var_guid_data2:x}"
            f"-{var_guid_data3:x}"
            f"-{var_guid_data4:x}"
            f"\nvar_name:       {var_name}"
            f"\nvar_data:       {var_data}"
        )
```

I tried adding support for it in UEFITool, but gave up because parsing this
format while maintaining compatibility is a bit of a mess thanks to the lack of
magic values.  Maybe someone else wants to do it, or I might get around to it
eventually.

It could also be that the dump is somehow altered or corrupted, but I only have
a single test machine at the moment and no other ways to dump the firmware.

NikolajSchlej, 2019-12-25:
It's not corrupted for sure, and it's indeed a new format for sending NVRAM contents from T2 side to x86 side on T2-equipped Macs, but I really doubt that a parser for it is needed to be implemented as a part of UEFITool (because that NVRAM block is not used by MacEFI as real NVRAM, just a seed for quicker initialization and PEI drivers).
Thanks for including the parser script though, appreciated.

closed 2019-12-25 by NikolajSchlej

kocoman1, 2022-04-01:
Is there any plan to support apple vss?

NikolajSchlej, 2022-04-01:
It should be already supported in new_engine branch. Do you have a file that it can't properly parse?

NikolajSchlej, 2022-04-01:
If you mean this particular compact format for T2 macs, the answer is "maybe later".

kocoman1, 2022-07-16:
I can open the NVRAM boot-args in NE, but in 0.28 it can't open .. just a huge file that I can't edit. I want to edit the boot-args to add like nv_disable=1 and -no_compat_check to the existing -v keepsyms=1 then reflash it.  others could edit the gpu-power-prefs to fix the "locked" nvram edits in linux