LongSoft/UEFITool #190: UEFIExtract causes memory corruption [new_engine]

hughsie, 2020-01-24:
I was doing `UEFIExtract infected.rom unpack` and was getting filenames like:

    CPD_entry_<C2>$_body.bin
    CPD_entry_<DF>$__body.bin
    CPD_entry_$DAT)_body.bin
    CPD_entry_$DAT)_info.txt
    CPD_entry_<C2>$_info.txt
    CPD_entry_<DF>$__info.txt
    CPD_entry_$<F3><89><DD>_<B3>_k<CF>'<D9>__body.bin
    CPD_entry_$<F3><89><DD>_<B3>_k<CF>'<D9>__info.txt
    CPD_entry_<FF><FF><FF><FF><FF><FF><FF><FF>$MCP_001_body.bin
    CPD_entry_<FF><FF><FF><FF><FF><FF><FF><FF>$MCP_001_info.txt

This looked like use-after free. Running UEFIExtract under valgrind gives me:

    ==398125== Invalid read of size 8
    ==398125==    at 0x430128: FfsParser::parseVendorHashFile(UByteArray const&, UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x43DC15: FfsParser::parseRawSectionBody(UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x4348EE: FfsParser::parseSectionBody(UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x434D1B: FfsParser::parseSections(UByteArray const&, UModelIndex const&, bool) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x435650: FfsParser::parseFileBody(UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x43609C: FfsParser::parseVolumeBody(UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x436D8E: FfsParser::parseRawArea(UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x43A451: FfsParser::parseBiosRegion(UByteArray const&, unsigned int, UModelIndex const&, UModelIndex&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x43BD9C: FfsParser::parseIntelImage(UByteArray const&, unsigned int, UModelIndex const&, UModelIndex&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x43D00F: FfsParser::performFirstPass(UByteArray const&, UModelIndex&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x43D1F9: FfsParser::parse(UByteArray const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x419E81: UEFIDumper::dump(UByteArray const&, Bstrlib::CBString const&, Bstrlib::CBString const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==  Address 0xb1ae4d0 is 32 bytes inside a block of size 81 free'd
    ==398125==    at 0x483AEFC: operator delete(void*) (vg_replace_malloc.c:586)
    ==398125==    by 0x430122: FfsParser::parseVendorHashFile(UByteArray const&, UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x43DC15: FfsParser::parseRawSectionBody(UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x4348EE: FfsParser::parseSectionBody(UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x434D1B: FfsParser::parseSections(UByteArray const&, UModelIndex const&, bool) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x435650: FfsParser::parseFileBody(UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x43609C: FfsParser::parseVolumeBody(UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x436D8E: FfsParser::parseRawArea(UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x43A451: FfsParser::parseBiosRegion(UByteArray const&, unsigned int, UModelIndex const&, UModelIndex&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x43BD9C: FfsParser::parseIntelImage(UByteArray const&, unsigned int, UModelIndex const&, UModelIndex&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x43D00F: FfsParser::performFirstPass(UByteArray const&, UModelIndex&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x43D1F9: FfsParser::parse(UByteArray const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==  Block was alloc'd at
    ==398125==    at 0x4839E86: operator new(unsigned long) (vg_replace_malloc.c:344)
    ==398125==    by 0x49BE74C: void std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_construct<char*>(char*, char*, std::forward_iterator_tag) (in /usr/lib64/libstdc++.so.6.0.27)
    ==398125==    by 0x440DE0: TreeModel::body(UModelIndex const&) const (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x430113: FfsParser::parseVendorHashFile(UByteArray const&, UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x43DC15: FfsParser::parseRawSectionBody(UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x4348EE: FfsParser::parseSectionBody(UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x434D1B: FfsParser::parseSections(UByteArray const&, UModelIndex const&, bool) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x435650: FfsParser::parseFileBody(UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x43609C: FfsParser::parseVolumeBody(UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x436D8E: FfsParser::parseRawArea(UModelIndex const&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x43A451: FfsParser::parseBiosRegion(UByteArray const&, unsigned int, UModelIndex const&, UModelIndex&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)
    ==398125==    by 0x43BD9C: FfsParser::parseIntelImage(UByteArray const&, unsigned int, UModelIndex const&, UModelIndex&) (in /home/hughsie/Code/UEFITool/UEFIExtract/UEFIExtract)

Which confirms this. I'm not a C++ developer, but I'm guessing this might be quite an easy fix with a little time. I'm afraid I can't share the `infected.rom` for obvious reasons :)

NikolajSchlej, 2020-01-24:
There are 2 issues indicated by this log:
1. ME parsing code needs support of MEv6-v10 structures, this is where CPD_entry_* comes from.
2. parseVendorHashFile has a UAF that needs to be found and fixed.

vit9696, 2020-01-24:
@hughsie could you run debug version of UEFIExtract under Clang's AddressSanitizer Valgrind output is less useful? Also, could you share infected.rom in private?

hughsie, 2020-01-24:
> could you run debug version of UEFIExtract under Clang's AddressSanitizer

[asan.txt](https://github.com/LongSoft/UEFITool/files/4110520/asan.txt)

> Also, could you share infected.rom in private

Unfortunately not, but I can see if I can make repack a non-infected FVH in the same way to reproduce the problem.

vit9696, 2020-01-24:
thanks for the report, but could yo build a debug version of UEFIExtract? This one neither has -g flag, nor -O0 instead of -O3/-Os or similar. Without this it does not give line numbers and filenames in the stack trace.

hughsie, 2020-01-24:
> This one neither has -g flag

My apologies. Something like this? 

[asan.txt](https://github.com/LongSoft/UEFITool/files/4110677/asan.txt)

vit9696, 2020-01-24:
Right, thanks. Will have a look when I get some spare time. Having a similar sample may help if you manage to craft it =)

vit9696, 2020-01-24:
Please check:

```
[new_engine 4d94847] Fix using temporary after scope in FfsParser::parseVendorHashFile
 1 file changed, 14 insertions(+), 10 deletions(-)
```

hughsie, 2020-01-24:
Better, but not perfect. No longer get a crash, but I do get:

    $ ls "infected.rom.dump/1 ME region/8 LOCL/0 CPD partition table"
    '0 '$'\004'                                                       '22 '$'\362\005''bmS'$'\277''?'$'\233\355\257\364''N'     '36 '
    '1 '$'\345'                                                       '23 ͊'$'\273''oe'$'\343\331\323''Ll*Y'                     '37 '$'\317\355''!~f'$'\236\306''r'$'\325''FDN'
    '10 y'$'\264''d_f'$'\340\235\026\276\002\243\346'                 '24 '$'\267\332''{'$'\357\277\024\356\r'']z'$'\375''B'    '38 '$'\001'
    '11 '$'\307\032''c'$'\226\340''g'$'\242''S'$'\237\217''J'$'\034'  '25 @'$'\276\370\307''g'$'\275'                           '39 '$'\224\204\271\205\356\313''ҵ'$'\301''#R.'
    '12 '$'\320''a'$'\240\214\300'')'$'\027\021\035''x'$'\327''T'     '26 '$'\005\021''H'$'\341\300''ҩW'$'\220''ԯ'$'\030'       '4 '
    '13 '$'\017\207''mc'$'\310''A'$'\235''@'$'\343''j'$'\a\n'         '27 '                                                     '40 '$'\215''m'$'\203\206\034\n''o'$'\250''i'$'\331\304''n'
    '14 d'$'\223\361'','$'\275''Ϗ'$'\020\312''g'$'\036\352'           '28 '                                                     '41 '$'\006\267'',+'$'\370\300\211''ç'$'\367\a\332'
    '15 '$'\025''e'$'\023''b'$'\346\331\033\325''F'$'\260\335''b'     '29  *'$'\257''wA2'$'\245\262\202\352''0'$'\362'          '5 @'
    '16 '$'\021'                                                      '3 '                                                      '6 :'$'\304''䄎'$'\372\273''cQs}'$'\273'
    '17 '$'\020\241''hY'$'\363'':'$'\032\021\314''knO'                '30 '$'\267\004'                                          '7 0'$'\226\346''3$'$'\005\200\354''ܬ'$'\t''S'
    '18 '$'\243\313\373\245\326''rS8'$'\006\277''['$'\210'            '31 '                                                     '8 -'$'\313''n'$'\363\372\340''K'$'\207\207'':'$'\277\233'
    '19 '$'\270''<5ܡ'$'\253\316''WoԮ'$'\342'                          '32 '$'\377\377\377\377\377\377\377\377\377\377\377\377'  '9 X'$'\005''F4'$'\266\326\325''8'$'\b\211\025\036'
    '2 '                                                              '33 '$'\377\377\377\377\377\377\377\377\377\377\377\377'   info.txt
    '20 '$'\317\021''в'$'\226''T'$'\375''_т'$'\260\372'               '34 '$'\377\377\377\377\377\377\377\377\377\377\377\377'
    '21 '$'\022\024''fJ'$'\237\036\330\311''Y'$'\025\351''g'          '35 '$'\377\377\377\377\377\377\377\377''$MCP'

Weirdly, nothing output from asan. I can run this under valgrind tomorrow if that would be helpful. I've also uploaded the ME partition, which should be clean :)

https://people.freedesktop.org/~hughsient/temp/Region_ME_ME_region.rgn

vit9696, 2020-01-24:
Well, there is no memory corruption anymore, just UEFIExtract does not (yet) understand this particular version of ME structures. We will handle it as time permits, but it may take some time as this is basically a feature request rather than a bugfix.

hughsie, 2020-01-24:
> does not (yet) understand this particular version of ME structures

Ahh, gotcha. I think in that case it might be better to not write the CPD files then, or at least give the files some kind of generated name.

NikolajSchlej, 2020-01-24:
Should we close this as complete and open an issue about incorrect CPD parsing for older ME versions instead?

vit9696, 2020-01-25:
Easier to just rename. Done.

vit9696, 2020-01-25:
For the time being I added a forgotten check for `$CPD` signature, so that we do not get garbage names at the very least.

[new_engine 1dccf3f] Fix CPD signature verification
 1 file changed, 1 insertion(+), 1 deletion(-)

hughsie, 2020-01-25:
That's great, thanks! It indeed fixes the garbage filenames. I don't actually need the new ME structures for the LVFS, so feel free to close this bug if the support isn't on your roadmap. Much appreciated for the speedy fix.

NikolajSchlej, 2020-01-25:
There is a vague plan for such support, but because ME6-10 images are outdated and not many people expressed interest in parsing them, this will probably be done very late, if ever. Closing for now, will reopen as a separate feature request later.

closed 2020-01-25 by NikolajSchlej