LongSoft/UEFITool #192: [UEFIPatch] Regions past EC aren't detected by old engine, which breaks patched images

NikolajSchlej, 2020-01-25:
Old engine's descriptor parsing code was written in 2014 and hasn't been improved much since then, so it can't see modern regions that aren't Descriptor/ME/GbE/BIOS/PDR/EC, and this breaks 
[C621D8A210.zip](https://github.com/LongSoft/UEFITool/files/4112222/C621D8A210.zip)
images like the attached one.

CrazyHulk, 2020-08-02:
Does lastest version have solve this?
I have find patch in patches.txt for C620.
`# PpmInitialize | 0xE2 & 0x1AA unlock for C620 Chipsets
3FFCAE95-23CF-4967-94F5-16352F68E43B 10 P:480BC2BA00200000:480BC2BA00000000 
3FFCAE95-23CF-4967-94F5-16352F68E43B 10 P:480BC2BA00800000:480BC2BA00000000`
Does it works for C621?

NikolajSchlej, 2020-08-03:
Even if t works, the resulting file will be broken due to regions past EC being missing after patching (the bug is about that).

CrazyHulk, 2022-08-08:
> Even if t works, the resulting file will be broken due to regions past EC being missing after patching (the bug is about that).

Sorry to bother you, is this feature still under development?

NikolajSchlej, 2022-08-09:
It is, albeit really slow due to me being on vacation. Will return to it when the time is right.

NikolajSchlej, 2022-09-13:
I'm mass-closing all issues that are specific to the old engine, because no further work is planned to be done on it, and all the bugs that are present there are to remain there forever. The original engine had a good run, but it needs to go in peace.

closed 2022-09-13 by NikolajSchlej

CrazyHulk, 2023-01-02:
> I'm mass-closing all issues that are specific to the old engine, because no further work is planned to be done on it, and all the bugs that are present there are to remain there forever. The original engine had a good run, but it needs to go in peace.

Does new engine can unlock 0xE2 & 0x1AA?