LongSoft/UEFITool #193: Add 3rd column to common/guids.csv

hughsie, 2020-01-27:
The LVFS decompresses thousands of firmware files. As part of that, we analyse the GUIDs of firmware, and try to present a "diff" of the different PEIMs/DXEs in an image. So that this makes more sense to the casual user, we map the GUID to not only the UI name (e.g. `Fat`) but also to a one-line description of the module, for instance:

     This UEFI driver detects the FAT file system in the disk. It also produces the Simple File System protocol for the consumer to perform file and directory operations on the disk. 

Although some of the descriptions are pretty bad (because they are auto-imported from the EDK2 .inf files) a lot of them have been manually added and I think there is some value in sharing them. Would a 3rd column in `guids.csv` be a good idea to store this kind of data? I understand if you're not terribly interested. :)

vit9696, 2020-01-27:
Hi, I can imagine something like this, but our current ideas with the database were mainly making it external and contained in some other repository. See: https://github.com/LongSoft/UEFITool/issues/177. Perhaps if you are interested in maintain something like that, we could work it out.

hughsie, 2020-01-27:
> Perhaps if you are interested in maintain something like that, we could work it out

I do think we need one place where we can all share protocol and DXE information. It's mad that a lot of projects like Ghidra/IDA all have 95% the same data with small differences. I'd happily help in any such initiative, the LVFS has tons of data I'd like to contribute.

vit9696, 2020-01-29:
Sorry for a slow reply. The above makes good sense to me, and I believe we should try to proceed with this, though even the code part from our end may be slow due to the lack of spare time. Not sure we can drive this. In any case, I believe we have the following list of questions to discuss:

- What should the database include? Currently it is just a set of GUIDs and their names for firmware files. What we can add here are GUIDs for protocols and other entities like UEFI variables or filesystem identifiers.
- What extra information should we provide? If we want to add protocols, for e.g. IDA it would be beneficial to also provide some headers or a link to headers for building plugins for reverse-engineering similar to ida-efitools. For your tool we may also need descriptions, because adding them to UEFITool, to be honest, I do not see how it can work in the GUI.
- How should we store it? I.e. what should the format/structure be, and how can we derive the files for different tools. E.g. how can we derive guids.csv for UEFITool to handle. I believe we should have some scripts in the repository.
- Where should we store it? Originally there was an idea to put it to github.com/acidanthera, but now that this basically failed, and we have people who actually need it, I wonder whether it is more reasonable to put it fwupdate.

hughsie, 2020-01-30:
> What should the database include

I think we need to identify if the GUID is a protocol or a file at least. i.e. a 'kind'

> a link to headers

I think that makes sense.

> How should we store it

A CSV file is probably the lowest-common-denominator.

> Where should we store it

Well, if you're open to the idea of not putting it in this repo, then we could put the "source" anywhere, perhaps somewhere easier to modify than a csv file in a repo. Perhaps we could even use wikipedia or a wiki like that, and then just "export" to CSV?

vit9696, 2020-02-02:
CSV lacks structure in it, which is problematic for 1 to many mappings. For example, 1 guid can have different names. I would rather go with something like yaml.

As for wiki, I would say it should rather be some git repository, as wiki's are hard to export and even harder to structure. It will also be a problem due to size, many browsers fail to render large pages.

But other than that it sounds good to me.

andrew-phi, 2020-02-05:
Wiki here is pullable as easy as any other git repository, just git clone https://github.com/orgname/somename.wiki.git
And please, not YAML *tears*

NikolajSchlej, 2022-08-24:
I really do not think we'll be doing this, at least certainly not in UEFITool repo. A centralised supported database of UEFI-related GUIDs would be a nice idea, even if those definitions and notes are to be imported from EDK2 or populated by reverse engineering, but it's certainly outside of the scope for this project that is currently on life support.

closed 2022-08-24 by NikolajSchlej

hughsie, 2022-08-24:
I'd be open to host this on the LVFS, if anyone wants to help.