LongSoft/UEFITool #194: Apple MP61 EFI Microcode detection/extraction issue

platomav, 2020-02-07:
At the attached Apple EFI scap image, GUID 153D2197-29BD-44DC-AC59-887F70E41A6B, only 1 microcode is detected/shown instead of 4. Extracting the "body" creates a file with the 1st microcode without its Header (not sure if this is expected behavior), thus not usable. Even if cutting off the Header is expected at "body" extraction, it still leaves the improper detection issue.

[image]

[MP61_F000_B00_2019-12-18_17-58_899394CB.zip](https://github.com/LongSoft/UEFITool/files/4169937/MP61_F000_B00_2019-12-18_17-58_899394CB.zip)

NikolajSchlej, 2020-02-07:
You can check if a header is expected to be cut by "Extract Body" by checking the header size field, which is 0x30 on the screenshot, so it's expected from the mechanical standpoint. However, a microcode binaries make very little sense without a header, I'll move them fully into the body, so both "Extract As Is" and "Extract Body" will return the same binary.
[ ]- Don't put microcode header into TreeItem.header

platomav, 2020-02-07:
I fully agree. There's also the detection issue (or lack thereof to be precise).

platomav, 2020-02-08:
Can confirm that 6fdc694 fixes both issues. All microcodes are now shown and both "Extract as is" and "Extract body" result in the same proper binary. 👍

commit 6fdc6941 (2020-02-08): Fix microcode detection

closes #194
closed 2020-02-08 by platomav