LongSoft/UEFITool #196: uefiextract: libc++abi.dylib: terminating with uncaught exception of type std::out_of_range: basic_string

marcomafcorp, 2020-02-24:
I continue to run into this bug trying to extract this binary with uefiextract. I'm not sure why this is happening.
[image]

NikolajSchlej, 2020-02-24:
Some byte array manipulations are using an incorrect (too big or negative) index. Please provide a binary that triggers it, so the issue could be reproduced.

hughsie, 2020-04-02:
The LVFS is hitting this too on 4 different capsule files from the same OEM. The firmware is all currently non-public, but I will send it to you via email. The backtrace looks like this:

    #0  0x00007ffff7c05a25 in raise () from /lib64/libc.so.6
    #1  0x00007ffff7bee895 in abort () from /lib64/libc.so.6
    #2  0x00007ffff7e4c961 in __gnu_cxx::__verbose_terminate_handler () at ../../../../libstdc++-v3/libsupc++/vterminate.cc:95
    #3  0x00007ffff7e5842c in __cxxabiv1::__terminate (handler=<optimized out>) at ../../../../libstdc++-v3/libsupc++/eh_terminate.cc:48
    #4  0x00007ffff7e58497 in std::terminate () at ../../../../libstdc++-v3/libsupc++/eh_terminate.cc:58
    #5  0x00007ffff7e58749 in __cxxabiv1::__cxa_throw (obj=obj@entry=0x55555b8c82e0, tinfo=0x7ffff7f90228 <typeinfo for std::out_of_range>, 
        dest=0x7ffff7e6f460 <std::out_of_range::~out_of_range()>) at ../../../../libstdc++-v3/libsupc++/eh_throw.cc:95
    #6  0x00007ffff7e4f44d in std::__throw_out_of_range_fmt (__fmt=__fmt@entry=0x5555555e0e98 "%s: __pos (which is %zu) > this->size() (which is %zu)")
        at ../../../../../libstdc++-v3/src/c++11/functexcept.cc:82
    #7  0x000055555558e6a2 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_check (__s=0x5555555e0e7f "basic_string::substr", __pos=<optimized out>, 
        this=<optimized out>) at /usr/include/c++/10/bits/char_traits.h:365
    #8  std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::substr (__n=<optimized out>, __pos=<optimized out>, this=<optimized out>)
        at /usr/include/c++/10/bits/basic_string.h:2835
    #9  UByteArray::mid (this=<optimized out>, pos=<optimized out>, len=<optimized out>) at /usr/src/debug/UEFITool-20200402-0.57.x86_64/common/bstrlib/../ubytearray.h:68
    #10 0x00005555555b6130 in FfsParser::checkProtectedRanges (this=0x7fffffffcfb0, index=...) at /usr/src/debug/UEFITool-20200402-0.57.x86_64/common/ffsparser.cpp:3367
    #11 0x00005555555b769b in FfsParser::performSecondPass (this=0x7fffffffcfb0, index=...) at /usr/src/debug/UEFITool-20200402-0.57.x86_64/common/ffsparser.cpp:3210
    #12 0x00005555555b8bef in FfsParser::parse (this=0x7fffffffcfb0, buffer=...) at /usr/src/debug/UEFITool-20200402-0.57.x86_64/common/ffsparser.cpp:127
    #13 0x000055555556e195 in main (argc=2, argv=0x7fffffffd438) at /usr/src/debug/UEFITool-20200402-0.57.x86_64/UEFIExtract/uefiextract_main.cpp:56

hughsie, 2020-04-21:
> The backtrace looks like this

If this is tricky to fix properly, I'm happy to fall back to `chipsec uefi decode` for those firmware files. I just need to make a plan if this is going to be tough to fix. Thanks, and sorry if I seem like nagging.

vit9696, 2020-04-21:
I completely forgot about it. Could you send it to vit9696 at protonmail?

hughsie, 2020-04-22:
@vit9696 ahh, looking back at my pile of email, I see this from @NikolajSchlej `However, the bug is there, and I'll fix it.` -- I can send you the firmware too, but I don't want to duplicate work if Nikolaj has already fixed the issue :)

vit9696, 2020-04-22:
I will ping him, but I am afraid Nikolaj is currently quite busy in Apple, so it might be easier for me to handle it if it is not something huge. Will not hurt to see a copy anyway.

hughsie, 2020-04-22:
@vit9696 you should have email :) Thanks.

NikolajSchlej, 2020-04-22:
I'm indeed busy as hell now, but the bug is basically a combination of the lack of boundary checks before calling UByteArray.mid() in various places and a garbed input due to it being an AMI PFAT capsule, not a real UEFI image.

vit9696, 2020-04-22:
Right. I was able to reproduce it, will see what I can do.

vit9696, 2020-04-22:
We will need to address it properly throughout the code (as well as understand what is going on with BootGuard in this firmware), but the crash-fix to the exact case is in new_engine branch.

hughsie, 2020-04-22:
It fixes the crash indeed on all 6 affected firmware files on the LVFS. Thanks!

NikolajSchlej, 2020-11-23:
Duplicate of #219, that will be used to rewrite that code eventually.

closed 2020-11-23 by NikolajSchlej