LongSoft/UEFITool #197: Module header seemingly corrupted when rebuilding image

HashFail, 2020-03-04:
Hi, 

I'm not sure if this is a bug or if something about this image is weird. When I modify the attached image, one of the module headers seems to be corrupted and all subsequent modules get smushed into one. To see what I am talking about, do the following: 

1) open `dxe-ffsv2` from the zip file
2) Replace module `E8F6A75C-3CDA-4B00-9837-8CA2A1F34EAC` with `SpsDxeModified` from the zip file (replace as is). The module/contents are not important, the file is simply the module with one byte changed
3) Save and open the new file. Where module `00310000-002E-0030-0000-FFFFFFFFFFFF` used to be, there is a module with a different GUID and there are no subsequent modules. There is also a warning about this module being unaligned.

There are also several warnings about invalid checksums and dependency expressions, including the module that gets mangled, but this doesn't happen to any other modules. I'm hoping someone can explain why this is happening, and in general it would be helpful if this sort of automatic correction were logged somewhere so users understand why their image is being changed.

Files: [uefitool-example.zip](https://github.com/LongSoft/UEFITool/files/4289971/uefitool-example.zip)

NikolajSchlej, 2020-03-05:
That original file looks severely corrupted during dumping, so UEFITool can't rebuild it properly. 
Can you point to the HP BIOS update with this or similar firmware, so we can check if it's a corruption or a curious file built this way by HP?

HashFail, 2020-03-05:
HP does not host BIOS updates publicly. The above is a single volume extracted from the image. I can provide the whole image to you directly by email, but please keep it confidential. 

I know that image appears corrupted. I'm surprised the image even boots. However, the full image contains a crypto signature that breaks if you modify a single byte, so I believe this the "correct" image.

To read it properly, I had to manually change the type/size fields in several modules/sections. I suspect that to modify it properly some similar edit needs to be made, but I can't figure out what's wrong. Previously, the image would fail to parse which let me know where to look, but now there's no module that fails to parse, only checksum/depex warnings.

Let me know where to send the image, and thank you for not only the help but all the work you do to provide this tool!

NikolajSchlej, 2020-03-05:
> HP does not host BIOS updates publicly. 
Can't be true, I had several HP systems in the past, and all of them had public BIOS update downloadable from HP support web site. Please provide your full HW model, I'll do the rest.

The fact it boots doesn't mean it's "correct" in any way. The machine might boot even with half of it's DXE volume is corrupted, if non of the corrupted drivers are essential for HW initialization or published required protocols.

Right now, you don't need to change anything yet. The first thing that needs to be done is to verify that the original image actually has this broken format.

HashFail, 2020-03-05:
It is true unfortunately, I believe they changed this a few years ago. The model is Proliant DL380 Generation 10. 

I don't think that the fact that it boots means it's "correct," but rather the fact that the signature verification works. The full update image contains some kind of private key signature over the whole image that verifies correctly when updating through the IPMI web interface.

NikolajSchlej, 2020-03-06:
Downloadable just fine from here: https://support.hpe.com/hpsc/swd/public/detail?swItemId=MTX-b1ffc5df57db42a1b558565848

The problem with that DXE volume is that it's compressed with a very uncommon LZMAF86 algorithm, and those issues are most likely caused by decompression failure in both UEFITool and UEFITool NE because both are treating it as normal LZMA. I'll investigate further.

NikolajSchlej, 2020-03-06:
Verified both engines to not support LZMAF86 properly right now, needs fixing.

HashFail, 2020-03-06:
Ah, thank you! I need support for the old engine so I will start looking into that.

vit9696, 2020-03-06:
@HashFail, @NikolajSchlej I added implementation for NE, you can add it to legacy if necessary.

closed 2020-03-15 by NikolajSchlej