LongSoft/UEFITool #205: Impossible to read HP BIOS?

mammamiamarcello, 2020-03-18:
I'm trying to extract the BIOS out of a Hp pavilion 15 cc112tu. I've downloaded the BIOS upgrade package from here (https://ftp.hp.com/pub/softpaq/sp101001-101500/sp101124.exe). I've unzipped the sp101124.exe file and and it created a InsydeFlash.exe. After extracing it again it created some files, among which 08361F33.bin. Opening it with UEFITool isn't working. Any fix?

vit9696, 2020-03-20:
The file you mentioned is either compressed or encrypted (most likely given files like CrisisKeyDLL), it does not contain an actual flashed UEFI image. Decrypting this file is out of the scope of UEFITool, so I suggest you find/write another tool that will perform this before feeding the file to UEFITool.

closed 2020-03-20 by vit9696

vit9696, 2020-03-20:
Apparently the tools can decrypt the firmware themselves. Editing `platform.ini` file in `[FDFile]` section to contain `FileName=08361F33.fd` and then running `HPFlashWinx32.exe` tool and choosing `Copy` produces `08361.bin`, which can then be opened just fine by UEFITool.

mammamiamarcello, 2020-03-20:
Yeah, when I wrote to you I magaed to get this file. My question was: can I extract this .bin file inside a Unix environment such as Linux/MacOS!

vit9696, 2020-03-20:
Well, if you understand the algorithm and write a cross-platform tool that does this, sure you can. But it is not a task for UEFITool.

dreamwhite, 2020-03-20:
Well, I've """found""" the algorithm. The steps are described in your comment:

> Apparently the tools can decrypt the firmware themselves. Editing `platform.ini` file in `[FDFile]` section to contain `FileName=08361F33.fd` and then running `HPFlashWinx32.exe` tool and choosing `Copy` produces `08361.bin`, which can then be opened just fine by UEFITool.

I'm trying to find the right solution as, for old Insyde BIOSes there was a @LongSoft utility called InsydeImageExtractor, but it's not working for modern HP BIOSes. Hope to get it working :)

dreamwhite, 2020-03-20:
Apparently a first attempt was done with https://github.com/Stane1983/HP_BIOS-extract-tool.
Currently testing it on a Mac environment, without success

dannyp777, 2024-08-03:
> Apparently a first attempt was done with https://github.com/Stane1983/HP_BIOS-extract-tool. Currently testing it on a Mac environment, without success

Only extracts the files from the softpak, doesn't decrypt the fd files. For sp88646 I extracted the archive on Linux, copied into a Win10 VM. Then ran the InsydeFlash.exe program and selected the third option to export the files to folder, and it then decrypted the fd files to bin + sig file. The reason I extracted the softpak file on Linux instead of running it in WIndows, is that it kept freezing up my VM, I don't know why.