LongSoft/UEFITool #208: UF/UE bug

LSSoniX, 2020-05-08:
Всем доброго здравия!

Посмотрите пожалуйста этот файл
https://ufile.io/shbe696p
UEFIFind и Extract крэшатся на нем даже на простых запроосах типа uefiextract <файл биоса>
UEFITool'ы открывают без проблем
UEFIReplace пока не проверял.
Спасибо.

vit9696, 2020-05-08:
Привет!

У меня всё ок на ветке из мастера, я отправил на сборку последние версии вот сюда: https://github.com/LongSoft/UEFITool/releases/tag/t20200508. Возможно проблема была исправлена.

Вит

LSSoniX, 2020-05-08:
Да, эти сборки исправились. Спасибо! :))

А была вот такая ерунда
[image]

vit9696, 2020-05-08:
Отлично, тогда через какое-то время выложу новую версию.

closed 2020-05-08 by vit9696