LongSoft/UEFITool #209: UEFIFind - new option

LSSoniX, 2020-05-31:
Всем доброго здравия!

На данный момент ключи для поиска
<header | body | all> <list | count>
Пробелм с этим нет. Но очень часто возникает запрос на получение номера типа Секции, тк невозможно понять под какой секцией лежит тот или иной файл. Ну если только в УТ смотреть.

Можно расширить ключ <list>?
То есть сделать 2 ключа.
Первый ключ <list> оставить как есть, на получение только GUID/SubGUID
Добавить второй, например, <list_extended>, по сути тот же <list>, но кроме GUID'ов выводил еще и номер секцтт. 

Примерно так:
uefifind bios.bin body list_extended <any pattern>
В пезультате:
365C62BA-05EF-4B2E-A7F7-92C1781AF4F9 19
Найдено в таком то GUID под секцией RAW
или
C02CFCE2-3021-42E6-8186-65FF0F5D9DE2 9F3A0016-AE55-4288-829D-D22FD344C347 18
Найдено в GUID/SubGUID по секцтей Freeform

Тем самым будет проще в скриптах для УЕ и УР, не надо будет задавать явные о секции. Тем более они не всегда постоянные - сегодня RAW,, завтра Freeform или Defined.

Спасибо.
PS названия ключей можно и другие

vit9696, 2020-05-31:
Привет, не обещаю, что сделаю прямо скоро, но тем не менее. Дай, пожалуйста, пару примеров прошивок, предполагаемых команд и ожидаемых результатов, чтобы мне ничего не напутать.

LSSoniX, 2020-06-07:
Всем доброго здра

@vit9696 
"Скоро только кошки родятся" О,Бендер :))
Извини, что долго с ответом. В общем архивов у меня пока мало, но на одном примере могу показать какая потрбность. 
И так. У нас есть бивис https://mega.nz/file/pfwXkTBa#z8tfQVrxoxAhubI_oKGvSw2FD4vA58KwWxpPQwHt0s0
Нам надо извлечь один файл, что-то поменять в нем и вернуть обратно
Ищем паттерн 24506E500102
`uefifind bios.bin body list 24506E500102`
gполучаем GUID A062CF1F-8473-4AA3-8793-600BC4FFE9A8
извлекаем
`uefiextract bios.bin A062CF1F-8473-4AA3-8793-600BC4FFE9A8 -o trmp -m body`
в резкльтате мы видим несколько файлов, а надо только один, то есть мы даже не знаем какой номер секции надо цказать в -t
Придется смотреть в УТ
[image]
это секция RAW, то есть для скриптов 19. При этом бываеи, что может оказать FreeForm (18) или еще чего похлеще.
Лады на раз нашли поправили и в УР, тем более номер секции мы посмотрели.

В общем всё хорошо для разовых действий в консоли, но в пакетной обработке, когда искомое может лежать в разных местах и в разных секцмях это тоска.
Поэтому давно появилась потрбность в дополнительном клюяе. По сути это тоже 'list', но чтобы помимо GUIDs выводилось и номер секции для -t УЕ и УР.
Примерно так:
`uefifind bios.bin body list_gs 24506E500102`
где ist_gs это новый ключ.
и в результате мы должны увидеть
A062CF1F-8473-4AA3-8793-600BC4FFE9A8 19

При этом сам list убирать не надо.
новый ключ оптимален только для body, для остальных он бесполезен.

Ну как то так.
Спасибо