LongSoft/UEFITool #210: Something wrong with my bios rom.

BoomAgain, 2020-07-08:
Hello, I use  AMI AFU for Aptio V to backup by bios file and I get a rom file.

But when I try to use uefi tool NE alpha 57 to open it. there are some info shown on the parser text panel.

the info are as follows . 

parseCpdRegion: CPD partition can't fit into it's region, truncated
parseCpdRegion: CPD partition is located outside of the opened image, skipped
parseCpdRegion: CPD partition is located outside of the opened image, skipped
parseCpdRegion: CPD partition is located outside of the opened image, skipped
parseRawArea: one of volumes inside overlaps the end of data
checkTeImageBase: TE image base is neither zero, nor original, nor adjusted, nor top-swapped
checkTeImageBase: TE image base is neither zero, nor original, nor adjusted, nor top-swapped
checkTeImageBase: TE image base is neither zero, nor original, nor adjusted, nor top-swapped
checkTeImageBase: TE image base is neither zero, nor original, nor adjusted, nor top-swapped
checkTeImageBase: TE image base is neither zero, nor original, nor adjusted, nor top-swapped
checkTeImageBase: TE image base is neither zero, nor original, nor adjusted, nor top-swapped
checkTeImageBase: TE image base is neither zero, nor original, nor adjusted, nor top-swapped
checkTeImageBase: TE image base is neither zero, nor original, nor adjusted, nor top-swapped
checkTeImageBase: TE image base is neither zero, nor original, nor adjusted, nor top-swapped
checkTeImageBase: TE image base is neither zero, nor original, nor adjusted, nor top-swapped
checkTeImageBase: TE image base is neither zero, nor original, nor adjusted, nor top-swapped
checkTeImageBase: TE image base is neither zero, nor original, nor adjusted, nor top-swapped
checkTeImageBase: TE image base is neither zero, nor original, nor adjusted, nor top-swapped
checkTeImageBase: TE image base is neither zero, nor original, nor adjusted, nor top-swapped
checkTeImageBase: TE image base is neither zero, nor original, nor adjusted, nor top-swapped
parseMeRegionBody: unknown ME region format

what should I do to parse it ?

and here is the bios file
 [amibios.rom](https://github.com/BoomAgain/tmp/raw/master/afuwin.rom)

NikolajSchlej, 2020-07-12:
Nothing is wrong with this file but the fact that it uses IFWI 2.0 layout that's currently not supported well enough by UEFITool NE. AMI decided to put invalid volume size into the OBB region, so it parses as Padding.

BoomAgain, 2020-07-13:
> Nothing is wrong with this file but the fact that it uses IFWI 2.0 layout that's currently not supported well enough by UEFITool NE. AMI decided to put invalid volume size into the OBB region, so it parses as Padding.

Thanks a lot. And Is there a good way  to valid it ?

NikolajSchlej, 2020-07-13:
What exactly are you trying to achieve with the changes you are picturing?

BoomAgain, 2020-07-18:
> What exactly are you trying to achieve with the changes you are picturing?

I tried to update some bios settings by flash a modified bios to motherboard, But faild. So...
I plan to change some bios settings by uefi shell. 
And  I want to know the setting  address in bios.

NikolajSchlej, 2020-07-20:
Attaching the DXE volume with fixed size, you can use the current UEFITool NE to extract the required data from it.
[dxe.bin.zip](https://github.com/LongSoft/UEFITool/files/4948914/dxe.bin.zip)

closed 2020-07-20 by NikolajSchlej