LongSoft/UEFITool #214: [patch] FPT parser: make ME8 bizzare name visible

nms42, 2020-07-29:
[me8-name.diff.txt](https://github.com/LongSoft/UEFITool/files/4992837/me8-name.diff.txt)

nms42, 2020-07-29:
The patch for new_engine.

NikolajSchlej, 2020-07-29:
Need more information on this. What is that bizzare name, where it comes from, is there a BIOS image with that name, etc.
The patch is trivial, but the story behind it is not.

nms42, 2020-07-29:
The name is \0xE0\x15\x00\x20. Unprintable by itself. That name is the bug in ME8 tools, probably. Attached the sample. The BIOS is for Toshiba Satellite L830 (almost legacy). I have seen that name in some other BIOSes too.

[bu8-bios-7.orig.fd.zip](https://github.com/LongSoft/UEFITool/files/4996187/bu8-bios-7.orig.fd.zip)

NikolajSchlej, 2020-07-31:
Need to improve it into a generic "display names as 8 hex digits if 4 char name has non-numeric-non-alphabet symbols". This will resolve this issue and all similar ones.

nms42, 2020-07-31:
And when there is a mixture of visible/invisible bytes? All to hex?

nms42, 2020-07-31:
Separate function in common/utility.cpp?

vit9696, 2020-07-31:
Yes, mixing ASCII and HEX will not do good for us. A function in utility.cpp sounds good to me.

nms42, 2020-08-02:
OK,
here is next attempt

[vh.diff.txt](https://github.com/LongSoft/UEFITool/files/5012222/vh.diff.txt)

NikolajSchlej, 2020-11-23:
Fixed in 115d338

closed 2020-11-23 by NikolajSchlej