LongSoft/UEFITool #216: [NE low priority enhancement] add support for FPT Header version 0x21 with crc32 checksum

nms42, 2020-07-29:
No patch yet

nms42, 2020-07-29:
Author of MEAnalyzer wrote to me: version 0x21 will be in firmware for TigerLake and following.

NikolajSchlej, 2020-07-29:
Need a BIOS image sample here before we go change the code.

nms42, 2020-07-29:
There are no BIOSes with 0x21 headers yet.

NikolajSchlej, 2020-07-29:
Let's wait for such a sample then before changing the code.

ValZapod, 2020-08-11:
There are samples in Intel leak. Crazy amount of them.

NikolajSchlej, 2022-08-25:
Found an image with FPT 0x21 header, uploading for future reference.
[ROG-MAXIMUS-Z690-HERO-ASUS-1720.CAP.zip](https://github.com/LongSoft/UEFITool/files/9421773/ROG-MAXIMUS-Z690-HERO-ASUS-1720.CAP.zip)

NikolajSchlej, 2022-08-25:
An image with FPT 0x20 header to test the other code branch
[MAXIMUS-IV-GENE-Z-GEN3-ASUS-3603.ROM.zip](https://github.com/LongSoft/UEFITool/files/9422214/MAXIMUS-IV-GENE-Z-GEN3-ASUS-3603.ROM.zip)
.

NikolajSchlej, 2022-08-25:
Resolved in A60, will be released soon.

closed 2022-08-25 by NikolajSchlej