LongSoft/UEFITool #217: QListWidget::itemDoubleClicked(QListWidgetItem*)’ is protected within this context

JonathonReinhart, 2020-07-30:
Environment:
- Debian 10
-  `sudo apt install qt4-qmake libqt4-dev-bin libqt4-dev`

```
$ ./unixbuild.sh
Attempting to build UEFITool NE for linux_x86_64...
Building UEFITool A57
g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -DU_ENABLE_FIT_PARSING_SUPPORT -DU_ENABLE_NVRAM_PARSING_SUPPORT -DU_ENABLE_ME_PARSING_SUPPORT -DU_ENABLE_GUID_DATABASE_SUPPORT -DQT_NO_DEBUG -DQT_GUI_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++-64 -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui -I/usr/include/qt4 -I. -I. -o uefitool.o uefitool.cpp
uefitool.cpp: In member function ‘virtual bool UEFITool::eventFilter(QObject*, QEvent*)’:
uefitool.cpp:765:65: error: ‘void QListWidget::itemDoubleClicked(QListWidgetItem*)’ is protected within this context
                 emit list->itemDoubleClicked(list->currentItem());
                                                                 ^
In file included from /usr/include/qt4/QtGui/QListWidget:1,
                 from uefitool.h:26,
                 from uefitool.cpp:15:
/usr/include/qt4/QtGui/qlistwidget.h:266:10: note: declared protected here
     void itemDoubleClicked(QListWidgetItem *item);
          ^~~~~~~~~~~~~~~~~
make: *** [Makefile:523: uefitool.o] Error 1
```

Git bisect points to https://github.com/LongSoft/UEFITool/commit/07dbfa3f964acb4217c3c90d52b80a475c6b7c49#diff-b6f9b660009c8e2a63edc71e30a3dfbcR765

```
git bisect start
# good: [2ef8d770e426e59115aa9fdd0ee942ba3617ae2e] Fix memory leak introduced in the previous commit
git bisect good 2ef8d770e426e59115aa9fdd0ee942ba3617ae2e
# bad: [fc2d8f07b101ac35cbe50ca1958bb00c89e62854] Version bump
git bisect bad fc2d8f07b101ac35cbe50ca1958bb00c89e62854
# good: [a3425115b8cd33e3ad5afbd6840000cdaa6d67fe] Add some more GUIDs to the database. (#199)
git bisect good a3425115b8cd33e3ad5afbd6840000cdaa6d67fe
# bad: [0f31b1401e2e522e7caab346e67014bb24489946] Print parent's name in search results (#201)
git bisect bad 0f31b1401e2e522e7caab346e67014bb24489946
# bad: [07dbfa3f964acb4217c3c90d52b80a475c6b7c49] Allow to use enter/return key within list widgets to navigate (#200)
git bisect bad 07dbfa3f964acb4217c3c90d52b80a475c6b7c49
# good: [ac1f24ec77010b0080267864a33ee903f16db050] Fix window is not in full screen mode after closing the app (#202)
git bisect good ac1f24ec77010b0080267864a33ee903f16db050
# first bad commit: [07dbfa3f964acb4217c3c90d52b80a475c6b7c49] Allow to use enter/return key within list widgets to navigate (#200)
```

vit9696, 2020-07-30:
Do you really need Qt4? Why not Qt5?

JonathonReinhart, 2020-07-30:
Well it's quite possible I don't know what I'm doing, but I couldn't figure out how to make it build with QT5. I have `qtchooser` installed:
```
QT_SELECT=5 ./unixbuild.sh 
Attempting to build UEFITool NE for linux_x86_64...
Building UEFITool A56
Project ERROR: Unknown module(s) in QT: core gui widgets
```

vit9696, 2020-07-30:
You will need to install these packages:
https://github.com/LongSoft/UEFITool/blob/new_engine/.travis.yml#L43

JonathonReinhart, 2020-07-30:
Okay I installed `qt5-default` and now I'm closer. Now I'm getting:
```
ui_searchdialog.h:13:10: fatal error: QtGui/QAction: No such file or directory
 #include <QtGui/QAction>
          ^~~~~~~~~~~~~~~
```

But that looks like a QT4-only file:
```
$ apt-file search -x 'QtGui/QAction$'
libqt4-dev: /usr/include/qt4/QtGui/QAction
```

Corroborated by:
- https://stackoverflow.com/a/21554163/119527

vit9696, 2020-10-19:
Qt4 is really too old and unsupported. Let's close this.

closed 2020-10-19 by vit9696