LongSoft/UEFITool #218: Customized decompression failed

hughsie, 2020-07-30:
Decompressing a P50 capsule on the LVFS from Lenovo:

    $ UEFIExtract S05KT61B.ROM
    parseVolumeHeader: unaligned volume
    parseVolumeHeader: unaligned volume
    parseVolumeHeader: unaligned volume
    parseVolumeHeader: unaligned volume
    parseVolumeNonUefiData: non-UEFI data found in volume's free space
    parseGuidedSectionBody: decompression failed with error Customized decompression failed
    parseVolumeBody: file header parsing failed with invalid size
    parse: not a single Volume Top File is found, the image may be corrupted

I get a few PEIMs in the decompression directory, but not the things I'm interested in. ROM available here: https://hughsie.com/temp/S05KT61B.ROM -- UEFIExtract report attached.

[S05KT61B.ROM.report.txt](https://github.com/LongSoft/UEFITool/files/5001214/S05KT61B.ROM.report.txt)

hughsie, 2020-07-30:
I forgot to say, but I'm using the latest git version from the `new_engine` branch.

NikolajSchlej, 2020-07-31:
This is once again not an UEFI image, but AMI PFAT capsule that [requires preprocessing](https://twitter.com/platomaniac/status/1051214985186172929) before submitting it to UEFIExtract.

NikolajSchlej, 2020-07-31:
That image unpacked by [ABGE v2.0](https://github.com/platomav/BIOSUtilities/releases/tag/ABGE-v2.0).
[S05KT61B.ROM_unpacked.bin.zip](https://github.com/LongSoft/UEFITool/files/5004996/S05KT61B.ROM_unpacked.bin.zip)

closed 2020-07-31 by NikolajSchlej