LongSoft/UEFITool #219: terminate called after throwing an instance of 'std::out_of_range'

hughsie, 2020-07-31:
Using the new_engine branch:

    terminate called after throwing an instance of 'std::out_of_range'
      what():  basic_string::substr: __pos (which is 18446744073698213888) > this->size() (which is 4194304)

    Program received signal SIGABRT, Aborted.
    0x00007ffff7ac59e5 in raise () from /lib64/libc.so.6
    (gdb) bt
    #0  0x00007ffff7ac59e5 in raise () from /lib64/libc.so.6
    #1  0x00007ffff7aae895 in abort () from /lib64/libc.so.6
    #2  0x00007ffff7e52961 in __gnu_cxx::__verbose_terminate_handler() [clone .cold] () from /lib64/libstdc++.so.6
    #3  0x00007ffff7e5e42c in __cxxabiv1::__terminate(void (*)()) () from /lib64/libstdc++.so.6
    #4  0x00007ffff7e5e497 in std::terminate() () from /lib64/libstdc++.so.6
    #5  0x00007ffff7e5e749 in __cxa_throw () from /lib64/libstdc++.so.6
    #6  0x00007ffff7e5544d in std::__throw_out_of_range_fmt(char const*, ...) [clone .cold] () from /lib64/libstdc++.so.6
    #7  0x00007ffff7efd9c8 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::substr(unsigned long, unsigned long) const () from /lib64/libstdc++.so.6
    #8  0x000000000042573d in UByteArray::mid(int, int) const ()
    #9  0x000000000042fae7 in FfsParser::checkProtectedRanges(UModelIndex const&) ()
    #10 0x0000000000432947 in FfsParser::performSecondPass(UModelIndex const&) ()
    #11 0x000000000043d594 in FfsParser::parse(UByteArray const&) ()
    #12 0x0000000000411f91 in main ()

When trying to extract this file: https://people.freedesktop.org/~hughsient/temp/BIOS_FV_BB.bin

Found on the LVFS... It could be the file isn't actually valid, but I'm pretty sure you don't want an exception.

NikolajSchlej, 2020-08-01:
Seriously need to harden that checkProtectedRanges() one, as we already had crashes in it, and the "fix" we had before is just silencing the crash.

vit9696, 2020-09-25:
Added a temporary workaround for now. Not closing as it needs thorough investigation. `addressDiff` looks apparently wrong for unclear reasons.

NikolajSchlej, 2022-08-24:
Will use this one to track the rewrite and inclusion of changes made in https://github.com/LongSoft/UEFITool/pull/225.

NikolajSchlej, 2022-09-08:
Turns out zero changes from that PR will end up integrated, and I've switched to Kaitai-based parsers for BGv2 structures instead. Uploading the files I've used for testing the parsers, so we can add them to the test suite later.
[BG2_ms_pmda_surface_laptop_2.cap.zip](https://github.com/LongSoft/UEFITool/files/9526269/ms_pmda_surface_laptop_2.cap.zip)
[BG2_GB_Z690AORUSMASTER.F8.zip](https://github.com/LongSoft/UEFITool/files/9526275/BG2_GB_Z690AORUSMASTER.F8.zip)
[BG2_lenovo_dump.bin.zip](https://github.com/LongSoft/UEFITool/files/9526276/gb2_lenovo_dump.bin.zip)

NikolajSchlej, 2022-09-09:
Files used to test new BGv1 parser, MS PMDA with v1 entries remains untested.
[l_bg1_1.bin.zip](https://github.com/LongSoft/UEFITool/files/9533325/l_bg1_1.bin.zip)
[lbg1_0.bin.zip](https://github.com/LongSoft/UEFITool/files/9533327/lbg1_0.bin.zip)
[g_bg1_0.bin.zip](https://github.com/LongSoft/UEFITool/files/9533332/g_bg1_0.bin.zip)
[surface laptop 1769-M1029273-001.bin.zip](https://github.com/LongSoft/UEFITool/files/9533333/surface.laptop.1769-M1029273-001.bin.zip)

NikolajSchlej, 2022-09-10:
Fixed in #288.

closed 2022-09-10 by NikolajSchlej