LongSoft/UEFITool #222: [Error] Fail to compile, collect2: error: ld returned 1 exit status

Direct-A, 2020-10-19:
I've cloned the `new_engine` branch, and run `./unixbuild.sh`.
Then I got the error below.
```
Attempting to build UEFITool NE for linux_x86_64...
Building UEFITool A57
g++ -Wl,-O2 -Wl,--sort-common -Wl,--as-needed -Wl,-z,relro -Wl,-z,now -Wl,--disable-new-dtags -Wl,--gc-sections -Wl,-rpath,/home/syc/miniconda3/lib -Wl,-rpath-link,/home/syc/miniconda3/lib -L/home/syc/miniconda3/lib -o UEFITool uefitool_main.o uefitool.o searchdialog.o hexviewdialog.o guidlineedit.o ffsfinder.o hexspinbox.o guiddatabase.o nvram.o nvramparser.o meparser.o ffsops.o types.o descriptor.o ffs.o peimage.o utility.o ffsbuilder.o ffsparser.o ffsreport.o ffsutils.o treeitem.o treemodel.o LzmaCompress.o LzmaDecompress.o Bra86.o LzFind.o LzmaDec.o LzmaEnc.o EfiTianoDecompress.o EfiTianoCompress.o EfiTianoCompressLegacy.o ustring.o sha256.o adler32.o compress.o crc32.o deflate.o gzclose.o gzlib.o gzread.o gzwrite.o inflate.o infback.o inftrees.o inffast.o trees.o uncompr.o zutil.o qhexedit.o chunks.o commands.o qrc_uefitool.o moc_uefitool.o moc_searchdialog.o moc_hexviewdialog.o moc_gotobasedialog.o moc_gotoaddressdialog.o moc_hexspinbox.o moc_treemodel.o moc_qhexedit.o moc_chunks.o moc_commands.o   /usr/lib/libQt5Widgets.so /usr/lib/libQt5Gui.so /usr/lib/libQt5Core.so -lGL -lpthread
/usr/bin/ld: /usr/lib/libQt5Widgets.so: undefined reference to `std::pmr::monotonic_buffer_resource::~monotonic_buffer_resource()@GLIBCXX_3.4.28'
/usr/bin/ld: /usr/lib/libQt5Widgets.so: undefined reference to `vtable for std::pmr::monotonic_buffer_resource@GLIBCXX_3.4.28'
collect2: error: ld returned 1 exit status
make: *** [Makefile:407: UEFITool] Error 1
```

I used 
```
OS: Manjaro Linux x86_64
Host: 20JAA007CD ThinkPad S5 2nd Gen
Kernel: 5.4.67-1-MANJARO
Uptime: 12 hours, 18 mins
Packages: 1295 (pacman), 5 (snap)
Shell: zsh 5.8
```

How can I solve this, to complate finish compile.

vit9696, 2020-10-19:
This looks like your Qt is built with c++17, but UEFITool is not. I believe it should work in either case and this is a bug of your distribution. As a workaround could you try adding -std=c++17 to the linkage command?

closed 2020-10-19 by vit9696

Direct-A, 2020-11-14:
Finally, I found out, it caused by install conda, compile command used path with `miniconda`. 
I've changed this part, and it could be complete compiled.