LongSoft/UEFITool #223: Вопрос по сообщению (UEFIRool/UEFIReplace v0.2x old engine)

LSSoniX, 2020-10-28:
Всем доброго здравия.

В последнее время на некоторых БИОСах, при замене файла в УТ/УР v0.2x, начало пояляться следующее сообщение:
'reconstructIntelImage: discovered unexpected 0.0 descriptor version, trying to continue...'
Что это и как к нему относиться?
Спасибо.

vit9696, 2020-10-28:
Привет, сам биос при этом работает? На каких это платформах? Можно пример?

dartraiden, 2020-10-28:
https://download.gigabyte.com/FileList/BIOS/mb_bios_z390-m-gaming_f9l.zip

Да, прошивка после этого работает.

NikolajSchlej, 2020-10-28:
```
        if (descriptorVersion == 2 && descriptorMap->DescriptorVersion != FLASH_DESCRIPTOR_VERSION_INVALID) {
            // Warn about incompatible version descriptor in case it is different.
            const FLASH_DESCRIPTOR_VERSION *version = (const FLASH_DESCRIPTOR_VERSION *)&descriptorMap->DescriptorVersion;
            if (version->Major != FLASH_DESCRIPTOR_VERSION_MAJOR || version->Minor != FLASH_DESCRIPTOR_VERSION_MINOR)
                msg(tr("reconstructIntelImage: discovered unexpected %1.%2 descriptor version, trying to continue...")
                    .arg(version->Major).arg(version->Minor));
        }
```

Ничего страшного, просто неизвестный формат дескриптора. Если патчи не трогают дескриптор, и границы регионов не смещены, можно игнорировать.

NikolajSchlej, 2022-08-25:
Ек думаю, что что-то нужно делать с этим всем, особенно в старом движке, но спасибо за репорт в любом случае, добавлю эти файлы в тестовую матрицу.

closed 2022-08-25 by NikolajSchlej

mirh, 2025-04-03:
> `unknown flash descriptor version 0.0`

FWIW this is how this looks today (for whoever may be googling the warning).