LongSoft/UEFITool #231: Asus pad-file issue

guub187, 2020-12-20:
Hi,
I'm trying to make a mod for an Asus Maximus V Gene board (latest bios 1903) from here:

[https://dlcdnets.asus.com/pub/ASUS/mb/LGA1155/MAXIMUS-V_GENE/MAXIMUS-V-GENE-ASUS-1903.zip](https://dlcdnets.asus.com/pub/ASUS/mb/LGA1155/MAXIMUS-V_GENE/MAXIMUS-V-GENE-ASUS-1903.zip)

But I've found an issue: when I open the bios using the A58(new engine), some "parseFileHeader: file alignment 10h is greater than parent volume alignment 8h" warnings appear... and a "pad-file" is generated just after the SandyBridgeGop DXE driver.

[image]

If I open it with the 0.28.0, no alignment warning appears, but I simply cannot remove any DXE driver or the bios will be incorrectly re-generated and the board won't boot.

[image]

However, if I remove the IntelSnbGopDriver, the "pad-file" dissapears and, then, I can remove some DXE drivers and everything works ok.

So I think you have a small bug there. I'm not sure but perhaps the IntelSnbGopDriver's code includes some kind of microcode-GUID chars inside "17088572-377F-44EF-8F4E-B09FFF46A070" and your UEFITool's 0.28.0/A58 parser thinks is a new microcode section when are just some GUID chars inside the DXE driver code...

Thanks.

NikolajSchlej, 2020-12-21:
The bug there is that Microcode file and all other files that are 16-bytes-aligned are not preserving the alignment of the file, and instead are using the alignment of the volume. This is not a bug in UEFITool, but in the ASUS image itself. You can enforce correct alignment manually by selecting all 5 pad-files (they are real, and used for aligning the following file) one by one and applying "Do Not Rebuild" action to all of them. If you do that, the resulting image will most likely continue to work.

NikolajSchlej, 2020-12-21:
Citing the UEFI PI spec:
"Alignment. The first byte of a volume is required to be aligned on some power-of-two boundary. At a minimum, this must be greater than or equal to the highest file alignment value."

UEFITool NE added a check for it after UEFITool 0.2x development had been done.

NikolajSchlej, 2020-12-21:
I stand corrected, the aligning function doesn't respect volume alignment, just uses hardcoded 8 byte default:
https://github.com/LongSoft/UEFITool/blob/44cafebbecb6485d183e8ed268d6d309682fe395/ffsengine.cpp#L3464

Reopening the bug to fix it.

guub187, 2020-12-21:
Ok, I'll test it when the patched 0.28.x version is released.
Many thanks!

82ghost82, 2020-12-30:
Add me to testers!
I think I have the same exact issue with a Asus Z9PE-D8 WS bios file (5802), pad file added on rebuild after modifying AmiBoardInfo, resulting in not bootable system.
Many thanks from my side too!

NikolajSchlej, 2022-09-13:
I'm mass-closing all issues that are specific to the old engine, because no further work is planned to be done on it, and all the bugs that are present there are to remain there forever. The original engine had a good run, but it needs to go in peace.

closed 2022-09-13 by NikolajSchlej

malavon, 2023-07-28:
I'm just here to tell you that this issue was closed incorrectly. The same errors show in NE A 67, at least in FreeBSD but I'm quite sure that's not related. It doesn't show in 0.28.0 though. The screenshots at the top of this issue confirm this, it's a problem in A 58 but not in 0.28.0. 
I'd say it's probably safe to say it's not part of the old engine, but rather the new one.

Cheers!

NikolajSchlej, 2023-07-28:
It doesn't show in 0.28.0 because there was no such check in 0.28.0. It's not an issue in UEFITool NE, but in the image itself, and UEFITool NE is correct to identify it and report on it. The issue with incorrect alignment during rebuild (that is indeed present in 0.28.0) is not present in NE because there's no rebuild code in it just yet.

malavon, 2023-07-28:
What I see in the image I have and the screenshots above, these are all messages about "parent volume alignment 8h". 

As far as I can see you identified the issue correctly before on 21 Dec 2020 - which I can indeed still see in the code -
where the alignment isn't read from the volume but hard coded to be 8h:
https://github.com/LongSoft/UEFITool/blob/44cafebbecb6485d183e8ed268d6d309682fe395/ffsengine.cpp#L3464-L3469

If I can I'll see if I can take a stab at this, but I'm currently trying to fix several bioses so I don't know if it'll be fast.

NikolajSchlej, 2023-07-28:
The thing is informational, and I haven't seen any real BIOS where alignment bits being set incorrectly actually lead to anything. It's more "stuff doesn't confirm to UEFI PI spec" than anything else, so you can happily ignore it.

malavon, 2023-07-29:
> The thing is informational, and I haven't seen any real BIOS where alignment bits being set incorrectly actually lead to anything. It's more "stuff doesn't confirm to UEFI PI spec" than anything else, so you can happily ignore it.

Hmm, good to know. Like I said I'm currently working on hacked/cracked Supermicro bioses that simply won't run, saw these errors and assumed they might be related. Sadly I can't verify this, I'm quite sure my internet connection is being injected with crap bioses each time I try to download them, even through https proxies. That or the Supermicro bioses do indeed have a lot of errors, e.g. the X10SLM+-LN4F bios I was trying to get to work.
I know this is a lot to ask, but could you maybe download that one and verify that it does indeed give these messages?
[Direct link from Supermicro](https://www.supermicro.com/Bios/softfiles/13392/x10slmp1_310.zip) or if that doesn't work [the support page](https://www.supermicro.com/en/support/resources/downloadcenter/firmware/MBD-X10SLM+-LN4F/BIOS).

NikolajSchlej, 2023-07-29:
[image]
Nothing to worry about. The files are indeed aligned correctly, but volume attributes don't reflect that alignment requirement. It's actually harmless for volumes that won't be copied into RAM one by one, and every volume in this particular BIOS is like that.

malavon, 2023-07-30:
Thank you! Good to know that this shouldn't be an issue. 

I'll take the opportunity to thank you for all of the work you've put into this and the other tools! I imagine you're probably one of the only developers considering this is a rather niche piece of software and I'm certain there's a lot of backend knowledge involved that doesn't really translate to the frontend. It's truly awesome and you probably don't hear it often enough!

malavon, 2023-07-31:
Ok, one last little request. Could you sha512 (or other hash function) the ROM-file you have? Just out of curiosity ;)

NikolajSchlej, 2023-08-01:
```
shasum -a 512 -b /Users/nschlej/Downloads/x10slmp1.310/x10slmp1.310
e5b491cb103e7c0b52d05ba6d90956acaaf3936d3adaa373ebbd6b88c6012e40aa3d18d236f8b0473249d9cfa92790e76270cd2bb44ad9b35d2a687797950a34
```

malavon, 2023-08-01:
Thanks! It is apparently the same, guess I was wrong there. I'll have to try and find out why it's not working when I modify the IME with a newer version (also out of SuperMicro bios). It must be something I'm overlooking, not so sure I'll ever be able to find it. Anyway, thank you again for everything!

e5b491cb103e7c0b52d05ba6d90956acaaf3936d3adaa373ebbd6b88c6012e40aa3d18d236f8b0473249d9cfa92790e76270cd2bb44ad9b35d2a687797950a34  x10slmp1.310

malavon, 2023-08-04:
Huh, that's odd ... in earlier screenshot with the errors you're missing 2 errors that I do see: "unable to parse AMI NVAR storage". Yet I tried the same UEFITool A67 (and A68) with different results. My problems might be stemming from there, and yet we do appear to have the exact same hash results. Suspicious and intriguing...
[image]
Sadly it wasn't enough to copy the first erroring section from a different bios (for a different motherboard with the same chipset).