LongSoft/UEFITool #232: BUG: PEI Core entry point can't be found

guub187, 2020-12-26:
Hi,

I'm trying to update microcodes for an old Asrock Z97M OC Formula board, latest bios v2.50 here:

[https://download.asrock.com/BIOS/1150/Z97M%20OC%20Formula(2.50)ROM.zip](https://download.asrock.com/BIOS/1150/Z97M%20OC%20Formula(2.50)ROM.zip)

but I get this error when the bios is rebuilt:

"PEI Core entry point can't be found in VTF. VTF is not patched"

Steps to repro:

1. Open UEFITool-0.28.0(old engine) and export the CPU microcode region (in FV#4, guid 17088572-377F-44EF-8F4E-B09FFF46A070)'s body to a file called, for instance "microcodes_body.raw"

2. Do nothing with the file, so it will be IDENTICAL to the original microdes!

3. Perform a "Replace body" with the extarcted "microcodes_body.raw" file and then "Save image..." so the bios is rebuilt...

4. Then, a "PEI Core VTF" error will appear... but we did NOTHING! We didn't patched anything! All we did was "Extract body..." + "Replace body..." with the same file we extracted !!!

The bios will be broken and won't boot, because UEFITool-0.28.0 reconstruct bad the bios. Test by yourself !!!

An screenshot:

[image]

Thanks.

NikolajSchlej, 2020-12-30:
>but we did NOTHING!
That's not true, as internally UEFITool performed a full volume rebuild, and that's definitely not a NOTHING.

>Test by yourself !!!
No need to get emotional. 

The issues with PEI volume rebuilds are known and there's very little we can do here without rewriting the build engine from scratch to fix it (that was the original goal of new_engine branch). I will leave this open until we can fix it, but the amount of work to do that is a lot, so I doubt it can be done soon.

guub187, 2020-12-31:
Many thanks!
By the way, do you have a paypal or address where to send you donations? oo

NikolajSchlej, 2022-09-13:
I'm mass-closing all issues that are specific to the old engine, because no further work is planned to be done on it, and all the bugs that are present there are to remain there forever. The original engine had a good run, but it needs to go in peace.

closed 2022-09-13 by NikolajSchlej