LongSoft/UEFITool #233: Общий вопрос по UEFITool's

LSSoniX, 2021-01-05:
Всем доброго здравия и с Новогодними праздниками.

БИОСы для материнок на базе АМД имеют размер 32 МЮ, по сути состоят из двух незвависмых БИОСов по 16 МБ. 
Есть в таких файлах какой либо признак определения что в файле содержатся 2 БИОС?
Я подозреваю что в природе могут существовать файлы с одним БИОС на 32 метра и как то не хотелось бы разделить пополам то что разделять не нужно.

Спасибо.

NikolajSchlej, 2021-01-06:
Нужны примеры и таких, и таких, иначе непонятно, как искать этот признак.

OZ1, 2022-09-12:
У intel с 11-го поколения тоже есть 32МБ биосы. В 600 Series Chipset PCH том 2 написано об этом: 7.1.9 Extended Bios Region

NikolajSchlej, 2022-09-13:
Это понятно, но на Intel нет никакой проблемы понять ни то, какая микросхема первая (та, которая начинается с дескриптора), ни то, одна там большая прошивка на 32 Мб, или две копии поменьше.

NikolajSchlej, 2025-11-18:
Done in https://github.com/LongSoft/UEFITool/pull/443, two 16 Mbit images are now detected as "AMD Image, Bank 0" and "AMD Image, Bank 1".