LongSoft/UEFITool #234: Ignore Intel Flash Descriptor dummies at AMD BIOS images

platomav, 2021-01-05:
HP seems to be using a dummy Intel Flash Descriptor structure at some AMD BIOS images, probably to satisfy their internal tools or something.

Obviously that causes false positives and errors/crashes at firmware parsers like UT so it should be ignored by adjusting the FD detection.

[image]

Checking that the 1st byte after the Signature (Flash Component Base Address, FCBA) is within the future-proof range of 01-10h should be enough. Future proof because ICH8-ICH10 = 1, IBX = 2 and CPT+ = 3.

[087FA_F02.zip](https://github.com/LongSoft/UEFITool/files/5772108/087FA_F02.zip)

NikolajSchlej, 2023-02-18:
We already have a check for a valid master base, the issue here is caused by incorrect fallback to raw area parsing in case Capsule or IntelDescriptor parsing failed. 
Fixed in https://github.com/LongSoft/UEFITool/commit/3a2072cfcbb50bd656372ca0cc2daf3d6dc39cbf

closed 2023-02-18 by NikolajSchlej

platomav, 2023-02-18:
Yes, it does seem to be working with the latest commits now:

[image]

NikolajSchlej, 2023-02-18:
Had to revert a fixing commit due to breakages introduces in the fuzzing-errors-fixing ones, will reapply.

NikolajSchlej, 2023-02-18:
Fixed again in https://github.com/LongSoft/UEFITool/commit/2d1ebcc11b9af03fa7ba779ebef80517cb1a3f65

platomav, 2023-02-18:
Indeed, can confirm that https://github.com/LongSoft/UEFITool/commit/2d1ebcc11b9af03fa7ba779ebef80517cb1a3f65 solves it.