LongSoft/UEFITool #238: 5Ch compress type not support

yarshure, 2021-05-27:
RT

vit9696, 2021-05-27:
Can you describe the issue properly and include the file?

yarshure, 2021-06-29:
[5CH Test file](https://github.com/yarshure/UEFITool/blob/master/6.bin)

NikolajSchlej, 2021-06-30:
It's not a compression type that isn't supported here, this image breaks EFI specification in multiple ways, but the worst offending change is that some structures are padded (by seems-to-be-random data) to 8 byte boundary. There's no good way to auto-detect this kind of spec breach, but I'll think about it.

yarshure, 2021-10-12:
@NikolajSchlej  The UEFI is iDevice diags, AAPL internal use,I just learn reverse it.

yarshure, 2021-11-06:
extract code use uefi_firmware
`# extract PE from iDevice diags(UEFI) section  body 
# sudo pip install uefi_firmware
#python e.py input output
import sys
from uefi_firmware import efi_compressor
handle=open(sys.argv[1], "rb")
content=handle.read()
newc=content[7:]
c=efi_compressor.TianoDecompress(newc,len(newc))
d=open(sys.argv[2], "wb")
st_mz=c.find(b"MZ")
d.write(c[st_mz:])
d.close()
handle.close()`

closed 2021-11-06 by yarshure