LongSoft/UEFITool #240: Several warnings when extracting BIOS from MSI Bravo 17

ydirson, 2021-07-11:
When using rev 34c8ad8dcc5f4bd76509123626d0f00511d3eb45 to extract the BIOS at https://www.msi.com/Laptop/support/Bravo-17-A4DX?sub_product=Bravo-17-A4DDK#down-bios I get a number of warnings

```
$ UEFIExtract E17FKAMS.117
parseGuidedSectionHeader: processing required bit set for GUIDed section with unknown GUID
parseSections: non-UEFI data found in sections area
parse: not a single Volume Top File is found, the image may be corrupted
```

I guess those point to new constructs and that some contents do not get extracted in the end ?

Specifically, I am looking for the video BIOS for the dGPU, and in the extracted files only the iGPU BIOS is to be found.

NikolajSchlej, 2021-07-14:
That image uses a PSP layout that UEFITool doesn't really support, and it's also too new for [psptool](https://github.com/PSPReverse/PSPTool) to understand. I doubt that volume that can't be decompressed has anything to do with dGPU though. Most likely it's on a separate I2C or SPI chip, but there aren't any good board photos to confirm or deny it.

booti386, 2021-12-18:
Yes, I can confirm that the vbios is on a separate spi chip.

closed 2022-08-24 by NikolajSchlej