LongSoft/UEFITool #243: Why there are no names of regions? Only GUIDs and Paddings

SpeedBird-C, 2021-09-10:
I'm pretty new to BIOS or UEFI stuff. So I'd like to get started studing it via mine UEFI of my laptop ASUS UX310UAK (on Intel)
So I've downloaded UEFI from official website and I have few questions:

- Insted of "Intel image" there is written UEFI image. Why? I have an Intel processor
- Insted of any resonable name like "ME region" or "BIOS region" there are just GUIDs and Paddings.Why? I want to see propper names to explore my UEFI
- Is it possible to see propper names insted GUIDs ?

See the screenshot below:
[image]

NikolajSchlej, 2021-09-11:
- the image you are opening is not an image of the full firmware, but only a BIOS region of it, that's why it's a "UEFI image", not "Intel image". 
- there are no regions in your file but BIOS region, and naming it "BIOS region" will be incorrect on AMD and other non-Intel boards.
- no, those volume GUIDs are mostly random and are only used to separate one volume from another when iterating over them. Files inside those volumes do have proper names, and some of their guids are non-random and known to us (i.e. listed in guids.csv)

closed 2021-09-11 by NikolajSchlej

SpeedBird-C, 2021-09-11:
> * the image you are opening is not an image of the full firmware, but only a BIOS region of it, that's why it's a "UEFI image", not "Intel image".
> * there are no regions in your file but BIOS [...]

Is it possible to work and explore this UEFI as any other propper UEFI? Would I find typical stuff of UEFIs ?
I cannot find so called guids.csv in this repository. Where is it located?