LongSoft/UEFITool #245: Display offsets / addresses of image entries like volumes, files, padding and others.

Ragebone, 2021-09-27:
The Information window contains useful information about the selected image entry. 
Including the "Full size" but not the address / offset at which the entry is located in the image. 
I would like to see that added to the information window.

As far as i can discern, it is neither displayed in the old nor new engine.

NikolajSchlej, 2021-09-28:
[image]
Base: offset from the beginning of the image file
Header address: physical address of the header (assuming the last byte of the image is at physical address 0xFFFFFFFF)
Data address: physical address of the data
Offset: offset from the beginning of the parent item

closed 2021-09-28 by NikolajSchlej

Ragebone, 2021-10-01:
My mistake, i appear to have failed in checking the new engine properly. 
It does indeed display those attributes.
The current Master branch does not, which appears to be the old engine in version 0.28.0.

NikolajSchlej, 2021-10-02:
Yes, current master doesn't have this function for multiple reasons, this is expected.