LongSoft/UEFITool #247: Compiling Warnings after #963671a

platomav, 2021-10-12:
The latest commits generate a lot of warnings when compiling under Windows 10 x64 using Qt v5.6.3 and MinGW v4.9.2. The guide followed is [this](https://github.com/LongSoft/UEFITool/wiki/How-to-Compile-UEFITool-for-Windows).

```
UEFITool-new_engine\common\nvramparser.cpp: In member function 'USTATUS NvramParser::parseNvarStore(const QModelIndex&)':
UEFITool-new_engine\common\nvramparser.cpp:350:72: warning: unknown conversion type character 'l' in format [-Wformat=]
                 info += usprintf("\nTimestamp: %" PRIX64 "h", timestamp);
                                                                        ^
UEFITool-new_engine\common\nvramparser.cpp:350:72: warning: too many arguments for format [-Wformat-extra-args]
UEFITool-new_engine\common\nvramparser.cpp: In member function 'USTATUS NvramParser::findNextStore(const QModelIndex&, const QByteArray&, UINT32, UINT32, UINT32&)':
UEFITool-new_engine\common\nvramparser.cpp:632:195: warning: unknown conversion type character 'l' in format [-Wformat=]
                     msg(usprintf("%s: FTW block candidate at offset %Xh skipped, has invalid body size %llXh", __FUNCTION__, localOffset + offset, (unsigned long long)ftw64Header->WriteQueueSize), index);
                                                                                                                                                                                                   ^
UEFITool-new_engine\common\nvramparser.cpp:632:195: warning: too many arguments for format [-Wformat-extra-args]
UEFITool-new_engine\common\nvramparser.cpp: In member function 'USTATUS NvramParser::parseVssStoreBody(const QModelIndex&, UINT8)':
UEFITool-new_engine\common\nvramparser.cpp:1538:96: warning: unknown conversion type character 'l' in format [-Wformat=]
             info += usprintf("\nMonotonic counter: %" PRIX64 "h\nTimestamp: ", monotonicCounter) + efiTimeToUString(timestamp)
                                                                                                ^
UEFITool-new_engine\common\nvramparser.cpp:1538:96: warning: too many arguments for format [-Wformat-extra-args]
UEFITool-new_engine\common\nvramparser.cpp: In member function 'USTATUS NvramParser::parseFlashMapBody(const QModelIndex&)':
UEFITool-new_engine\common\nvramparser.cpp:1928:36: warning: unknown conversion type character 'l' in format [-Wformat=]
                 entryHeader->Offset);
                                    ^
UEFITool-new_engine\common\nvramparser.cpp:1928:36: warning: format '%X' expects argument of type 'unsigned int', but argument 4 has type 'UINT64 {aka long long unsigned int}' [-Wformat=]
UEFITool-new_engine\common\nvramparser.cpp:1928:36: warning: too many arguments for format [-Wformat-extra-args]
UEFITool-new_engine\common\meparser.cpp: In member function 'USTATUS MeParser::parseFptRegion(const QByteArray&, const QModelIndex&, QModelIndex&)':
UEFITool-new_engine\common\meparser.cpp:180:38: warning: format '%lX' expects argument of type 'long unsigned int', but argument 2 has type 'unsigned int' [-Wformat=]
                         ptEntry->Type);
                                      ^
UEFITool-new_engine\common\meparser.cpp:180:38: warning: format '%lu' expects argument of type 'long unsigned int', but argument 3 has type 'unsigned int' [-Wformat=]
UEFITool-new_engine\common\meparser.cpp: In member function 'USTATUS MeParser::parseIfwi16Region(const QByteArray&, const QModelIndex&, QModelIndex&)':
UEFITool-new_engine\common\meparser.cpp:336:69: warning: unknown conversion type character 'l' in format [-Wformat=]
                             (unsigned long long)ifwiHeader->Checksum);
                                                                     ^
UEFITool-new_engine\common\meparser.cpp:336:69: warning: too many arguments for format [-Wformat-extra-args]
UEFITool-new_engine\common\utility.cpp: In function 'QString errorCodeToUString(USTATUS)':
UEFITool-new_engine\common\utility.cpp:187:93: warning: format '%lX' expects argument of type 'long unsigned int', but argument 2 has type 'USTATUS {aka unsigned int}' [-Wformat=]
     default:                                return usprintf("Unknown error %02lX", errorCode);
                                                                                             ^
UEFITool-new_engine\common\ffsparser.cpp: In member function 'USTATUS FfsParser::parsePeImageSectionBody(const QModelIndex&)':
UEFITool-new_engine\common\ffsparser.cpp:3167:42: warning: unknown conversion type character 'l' in format [-Wformat=]
             optionalHeader.H64->ImageBase);
                                          ^
UEFITool-new_engine\common\ffsparser.cpp:3167:42: warning: too many arguments for format [-Wformat-extra-args]
UEFITool-new_engine\common\ffsparser.cpp: In member function 'USTATUS FfsParser::parseTeImageSectionBody(const QModelIndex&)':
UEFITool-new_engine\common\ffsparser.cpp:3209:87: warning: unknown conversion type character 'l' in format [-Wformat=]
             teHeader->ImageBase + teHeader->StrippedSize - sizeof(EFI_IMAGE_TE_HEADER));
                                                                                       ^
UEFITool-new_engine\common\ffsparser.cpp:3209:87: warning: unknown conversion type character 'l' in format [-Wformat=]
UEFITool-new_engine\common\ffsparser.cpp:3209:87: warning: too many arguments for format [-Wformat-extra-args]
UEFITool-new_engine\common\ffsparser.cpp: In member function 'USTATUS FfsParser::addInfoRecursive(const QModelIndex&)':
UEFITool-new_engine\common\ffsparser.cpp:3377:115: warning: unknown conversion type character 'l' in format [-Wformat=]
                 model->addInfo(index, usprintf("Data address: %08llXh\n", (unsigned long long)address + headerSize),false);
                                                                                                                   ^
UEFITool-new_engine\common\ffsparser.cpp:3377:115: warning: too many arguments for format [-Wformat-extra-args]
UEFITool-new_engine\common\ffsparser.cpp:3378:104: warning: unknown conversion type character 'l' in format [-Wformat=]
                 model->addInfo(index, usprintf("Header address: %08llXh\n", (unsigned long long)address), false);
                                                                                                        ^
UEFITool-new_engine\common\ffsparser.cpp:3378:104: warning: too many arguments for format [-Wformat-extra-args]
UEFITool-new_engine\common\ffsparser.cpp:3381:97: warning: unknown conversion type character 'l' in format [-Wformat=]
                 model->addInfo(index, usprintf("Address: %08llXh\n", (unsigned long long)address), false);
                                                                                                 ^
UEFITool-new_engine\common\ffsparser.cpp:3381:97: warning: too many arguments for format [-Wformat-extra-args]
UEFITool-new_engine\common\ffsparser.cpp: In member function 'USTATUS FfsParser::parseFit(const QModelIndex&)':
UEFITool-new_engine\common\ffsparser.cpp:3862:83: warning: unknown conversion type character 'l' in format [-Wformat=]
         currentStrings.push_back(usprintf("%016" PRIX64 "h", currentEntry->Address));
                                                                                   ^
UEFITool-new_engine\common\ffsparser.cpp:3862:83: warning: too many arguments for format [-Wformat-extra-args]
UEFITool-new_engine\common\ffsparser.cpp: In member function 'USTATUS FfsParser::parseFitEntryAcm(const QByteArray&, UINT32, const QModelIndex&, QString&, UINT32&)':
UEFITool-new_engine\common\ffsparser.cpp:4013:25: warning: format '%lX' expects argument of type 'long unsigned int', but argument 5 has type 'unsigned int' [-Wformat=]
                         );
[... 14 lines cut ...]
                         ptEntry->Size) +
                                      ^
UEFITool-new_engine\common\ffsparser.cpp:4562:38: warning: format '%lu' expects argument of type 'long unsigned int', but argument 3 has type 'unsigned int' [-Wformat=]
UEFITool-new_engine\common\ffsparser.cpp: In member function 'USTATUS FfsParser::parseCpdRegion(const QByteArray&, UINT32, const QModelIndex&, QModelIndex&)':
UEFITool-new_engine\common\ffsparser.cpp:5015:134: warning: format '%lX' expects argument of type 'long unsigned int', but argument 3 has type 'unsigned int' [-Wformat=]
                                          manifestHeader->ExponentSize * sizeof(UINT32), manifestHeader->ExponentSize * sizeof(UINT32));
[... cut ...]
                                                                                                                                               ^
UEFITool-new_engine\common\ffsparser.cpp: In member function 'USTATUS FfsParser::parseSignedPackageInfoData(const QModelIndex&)':
UEFITool-new_engine\common\ffsparser.cpp:5253:164: warning: format '%lu' expects argument of type 'long unsigned int', but argument 4 has type 'unsigned int' [-Wformat=]
                 msg(usprintf("%s: CPD Signed Package Info hash size is %d, expected %lu", __FUNCTION__, moduleHeader->HashSize, sizeof (moduleHeader->MetadataHash)), extIndex);
```

NikolajSchlej, 2022-08-29:
Reproduced, with even more warnings regarding `struct x y = {}` initialisers, but none of those are fatal in any way, and I'd recommend not to waste time in trying to be compatible to an old GCC build for a non-native platform.

We are building Windows executables with MSVS 2017 using a deprecated v141_xp target, the exact steps can be obtained from CI\CD configuration here: https://github.com/LongSoft/UEFITool/blob/new_engine/.github/workflows/main.yml#L140

I will likely be somewhat beneficial to switch to a newer MinGW with GCC 9.x for those build-your-stuff-at-home folks, but that's out of scope for this issue, so I'll create a new one for it.

closed 2022-08-29 by NikolajSchlej

platomav, 2022-08-29:
Yeah, these versions are used by [this](https://github.com/LongSoft/UEFITool/wiki/How-to-Compile-UEFITool-for-Windows) wiki page I had created 3 years ago on how to compile under Windows. Just tested it again with the latest NE branch, and it still works despite the warnings. Alas, I completely understand your point and the CI\CD method should work more natively. We could keep the wiki page (since it still works) but if you want to delete it due to age, no problem. Thanks for looking into this, Nikolaj. ;)

NikolajSchlej, 2022-08-29:
No need to remove anything, as our unixbuild.sh script does the same thing for Windows anyway. Point is, the resulting image still works despite all the warnings, so everything is fine already and nothing needs to be deleted.