LongSoft/UEFITool #256: Crash when parsing malformed ROM file

retpoline, 2022-01-24:
Hi folks,

An interesting crash was found while fuzz testing of the **UEFIextract** binary which can be triggered via a malformed ROM file. Although this malformed file only crashes the program as-is, it could potentially be crafted further and create a security issue where these kinds of files would be able compromise the process's memory through taking advantage of affordances given by memory corruption. It's recommend to harden the code to prevent these kinds of bugs as it could greatly mitigate such this issue and even future bugs.

Crashing ROM repro file (~4mb): https://ufile.io/qqjn4c54

**debug log**

```
$ UEFIExtract crash.rom 
Segmentation fault (core dumped)

$ gdb -q UEFIExtract
Reading symbols from UEFIExtract...

(gdb) r crash.rom 
Starting program: UEFIExtract crash.rom

Program received signal SIGSEGV, Segmentation fault.
0x000055555557d99f in MeParser::parseFptRegion(UByteArray const&, UModelIndex const&, UModelIndex&) ()

(gdb) bt
#0  0x000055555557d99f in MeParser::parseFptRegion(UByteArray const&, UModelIndex const&, UModelIndex&) ()
#1  0x000055555557fa7b in MeParser::parseMeRegionBody(UModelIndex const&) ()
#2  0x0000555555581d23 in FfsParser::parseMeRegion(UByteArray const&, unsigned int, UModelIndex const&, UModelIndex&) ()
#3  0x0000555555595ecf in FfsParser::parseIntelImage(UByteArray const&, unsigned int, UModelIndex const&, UModelIndex&) ()
#4  0x0000555555597521 in FfsParser::performFirstPass(UByteArray const&, UModelIndex&) ()
#5  0x0000555555597730 in FfsParser::parse(UByteArray const&) ()
#6  0x00005555555665d5 in main ()

(gdb) i r
rax            0x0                 0
rbx            0x0                 0
rcx            0x190               400
rdx            0x0                 0
rsi            0x5555555f6d90      93824992898448
rdi            0x0                 0
rbp            0x7fffffffd930      0x7fffffffd930
rsp            0x7fffffffd720      0x7fffffffd720
r8             0x7fffffffd8c8      140737488345288
r9             0x7fffffffd800      140737488345088
r10            0x170               368
r11            0x7ffff7da7be0      140737351678944
r12            0x7fffffffd800      140737488345088
r13            0x7fffffffd7d3      140737488345043
r14            0x0                 0
r15            0x3f                63
rip            0x55555557d99f      0x55555557d99f <MeParser::parseFptRegion(UByteArray const&, UModelIndex const&, UModelIndex&)+1455>
eflags         0x10246             [ PF ZF IF RF ]
cs             0x33                51
ss             0x2b                43
ds             0x0                 0
es             0x0                 0
fs             0x0                 0
gs             0x0                 0

(gdb) x/i $rip
=> 0x55555557d99f <_ZN8MeParser14parseFptRegionERK10UByteArrayRK11UModelIndexRS3_+1455>:	cmp    %ecx,0x8(%rax)

(gdb) exploitable
Description: Access violation near NULL on destination operand
Short description: DestAvNearNull (15/22)
Hash: 8dc0ff2bd6c793bfca1c8a11b2822885.2eef4809031dd856ca2bd1a8795dcfee
Exploitability Classification: PROBABLY_EXPLOITABLE
Explanation: The target crashed on an access violation at an address matching the destination operand of the instruction. This likely indicates a write access violation, which means the attacker may control write address and/or value. However, it there is a chance it could be a NULL dereference.
Other tags: AccessViolation (21/22)
```

NikolajSchlej, 2022-01-27:
Reuploaded the file here for later investigation:
[crash.rom.zip](https://github.com/LongSoft/UEFITool/files/7947255/crash.rom.zip)

The ME parsing code had been a rushed development, and it's not the first crash in it by far. Will be fixed.

NikolajSchlej, 2022-09-13:
Investigated, the crash is a result of all entries in FPT partition table being marked as `Invalid`, and a following use of partitions.front() iterator on an empty set. Will be fixed in the next release.

NikolajSchlej, 2022-09-15:
Fixed by #156

closed 2022-09-15 by NikolajSchlej